h2zero/NimBLE-Arduino #367: Crash when connecting

I-Connect, 2022-03-09:
Hi, 

I am writing a lib that communicates with a Nuki lock.

The lock has a connection timeout so before sending a command I check if I am still connected and if not I reconnect.
Sometimes the esp32 crashes when connecting to the lock.

This is the code for connecting (full code: https://github.com/I-Connect/nuki_ble)

```
bool NukiBle::connectBle(BLEAddress bleAddress) {
  //TODO fix esp crash when connection fails or is interupted...?
  if (!pClient->isConnected()) {
    uint8_t connectRetry = 0;
    while (connectRetry < 5) {
      if (pClient->connect(bleAddress, true)) {
        if (pClient->isConnected() && registerOnGdioChar() && registerOnUsdioChar()) {  //doublecheck if is connected otherwise registiring gdio crashes esp
          bleConnected = true;
          return true;
        } else {
          log_w("BLE register on pairing or data Service/Char failed");
        }
      } else {
        log_w("BLE Connect failed");
      }
      connectRetry++;
      esp_task_wdt_reset();
      delay(200);
    }
  } else {
    return true;
  }
  return false;
}

```

I retry the connection 5 times, crash seems to happen when there is an "onConnect" event but somehow it is not really connected but then crashes on registering/writing?
It often happens immediately after uploading new code and the esp is rebooted but also during regular execution.

Maybe it is due to a bad connection, I often see multiple retries but at least I should be able to catch this and prevent the esp from crashing...

result logs:

```
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: Connection failed; status=574
[W][NukiBle.cpp:112] connectBle(): BLE Connect failed
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: Connection failed; status=574
[W][NukiBle.cpp:112] connectBle(): BLE Connect failed
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: Connection failed; status=574        
[W][NukiBle.cpp:112] connectBle(): BLE Connect failed
[D][NukiBle.cpp:1242] onConnect(): BLE connected
lld_pdu_get_tx_flush_nb HCI packet count mismatch (3, 4)
E NimBLEClient: characteristicDiscCB() rc=7
E NimBLEClient: Could not retrieve services
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d7812  PS      : 0x00060130  A0      : 0x800d7ae0  A1      : 0x3ffc6390
A2      : 0x00000000  A3      : 0x3ffc63f4  A4      : 0x2b5f96a6  A5      : 0x00005501  
A6      : 0x000011e4  A7      : 0xff000000  A8      : 0x800d90d4  A9      : 0x3ffc6380
A10     : 0x3ffc63f4  A11     : 0x3ffc63a8  A12     : 0x00000015  A13     : 0x3ffc6408  
A14     : 0x00000001  A15     : 0x2ee10155  SAR     : 0x00000018  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000004  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x400d7812:0x3ffc6390 0x400d7add:0x3ffc63d0 0x400d32da:0x3ffc6430 0x400d3496:0x3ffc6470 0x400d36dc:0x3ffc64a0 0x400d3779:0x3ffc6580 0x400d3d58:0x3ffc65a0 0x400d3e96:0x3ffc6630 0x400d4fa1:0x3ffc6730 0x400ec8dd:0x3ffc6750 0x4008fea2:0x3ffc6770
  #0  0x400d7812:0x3ffc6390 in NimBLERemoteService::getCharacteristic(NimBLEUUID const&) at .pio\libdeps\debug\NimBLE-Arduino\src/NimBLERemoteService.cpp:78
  #1  0x400d7add:0x3ffc63d0 in NimBLERemoteService::getCharacteristic(char const*) at .pio\libdeps\debug\NimBLE-Arduino\src/NimBLERemoteService.cpp:78
  #2  0x400d32da:0x3ffc6430 in NukiBle::registerOnGdioChar() at src/NukiUtills.h:6
  #3  0x400d3496:0x3ffc6470 in NukiBle::connectBle(NimBLEAddress) at src/NukiUtills.h:6
  #4  0x400d36dc:0x3ffc64a0 in NukiBle::sendEncryptedMessage(NukiCommand, unsigned char*, unsigned char) at src/NukiUtills.h:6
  #5  0x400d3779:0x3ffc6580 in NukiBle::cmdStateMachine(NukiBle::NukiAction) at src/NukiUtills.h:6
  #6  0x400d3d58:0x3ffc65a0 in NukiBle::executeAction(NukiBle::NukiAction) at src/NukiUtills.h:6
  #7  0x400d3e96:0x3ffc6630 in NukiBle::updateKeyTurnerState() at src/NukiUtills.h:6
  #8  0x400d4fa1:0x3ffc6730 in loop() at src/main.cpp:72
  #9  0x400ec8dd:0x3ffc6750 in loopTask(void*) at C:\Users\Jeroen\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:23
  #10 0x4008fea2:0x3ffc6770 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

could you give me a hand with this?
thx!
Jeroen

h2zero, 2022-03-10:
It looks like you've called a method on an unitialized characteristic in the function 'registerOnGdioChar()`. You should check if the call to `getCharacteristic()` returns a non-nullptr value before calling any methods on it.

I-Connect, 2022-03-10:
yup that was the issue, no crashes anymore.

THX!!

h2zero, 2022-04-04:
Closing as this has been resolved.

closed 2022-04-04 by h2zero