h2zero/NimBLE-Arduino #375: ESP32 S2 fails to compile

joostbijl, 2022-03-23:
Hi, i'm having an issue compiling for ESP32 S2 in Arduino IDE.

I'm using the ESP32-BLE-Keyboard library that, as of recently, supports NimBLE. I can succesfully compile and run code on regular 'devkit' for ESP32 (S??) and on ttgo t7 1.3. However, if i want to compile for 'devkit s2' so i can upload it to wemos S2 mini, the compiler throws an error (see below). Any thoughts on what i'm doing wrong here?

- NimBLE version 1.3.7
- ESP32 libraries v 2.0.2
- BleKeyboard (recen RC)

Error i get is:

`
Multiple libraries were found for "WiFi.h"
C:\Users\JoostBijl\Onedrive\Documents\Arduino\libraries\NimBLE-Arduino\src\esp-hci\src\esp_nimble_hci.c:30:10: fatal error: esp_bt.h: No such file or directory
 Used: C:\Users\JoostBijl\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\WiFi
 #include "esp_bt.h"
 Not used: C:\Users\JoostBijl\Onedrive\Documents\Arduino\libraries\WiFiNINA
          ^~~~~~~~~~
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
compilation terminated.
exit status 1
Error compiling for board ESP32S2 Dev Module.
`

thorrak, 2022-03-23:
Doesn’t the S2 not have a Bluetooth radio?

joostbijl, 2022-03-23:
doh, it does not :( !

That's a bit silly from my end. Sorry for the inconvenience

closed 2022-03-23 by joostbijl

thorrak, 2022-03-23:
For what it's worth, the LoLin ESP32-C3 Mini has fewer pins than the S2, but includes a Bluetooth radio. That said - I was testing one out yesterday and unfortunately couldn't get it working in Platformio. 

Depending on your application, it may be worth a look, but it does feel like it is still a "work in progress".