h2zero/NimBLE-Arduino #377: GATT_INSUF_RESOURCE when trying to write data to characteristic

qdlmcfresh, 2022-03-24:
Im trying to send about 300 to 400 bytes to a characteristic but get error GATT_INSUF_RESOURCE (0x11). Reading the same amount of data from the same characteristic works though

h2zero, 2022-03-24:
That's a return code from the peer device usually. What is the device you're sending the bytes to?

qdlmcfresh, 2022-03-25:
Im sending from an android phone to an esp32

h2zero, 2022-03-25:
Oh okay, in that case would it be possible to provide a log from the esp32 when this occurs? My initial thought is this is most likely being sent as a long write, which could cause this due to limited buffers. You could try increasing [this value](https://github.com/h2zero/NimBLE-Arduino/blob/c0d1ce06fc34de72f88cdfdd4ef577d0e783a40c/src/nimconfig.h#L83)

qdlmcfresh, 2022-03-25:
Your first thought seems to be right. increasing CONFIG_BT_NIMBLE_MSYS1_BLOCK_COUNT works. Thank you :)

closed 2022-03-25 by qdlmcfresh