h2zero/NimBLE-Arduino #378: pBLEClient->getCharacteristic(handle) returns NullPtr

miksumin, 2022-04-03:
Hello, I found that using pBLEClient->getCharacteristic(handle) right after successful connection returns NullPtr.
After debugging the issue I found that discovering of Services and Characteristics are missing in code.
So I suggest to add it:
NimBLERemoteCharacteristic* NimBLEClient::getCharacteristic(const uint16_t handle)
{
	
	**getServices(true);**			// added
	
	NimBLERemoteService *pService = nullptr;
    for(auto it = m_servicesVector.begin(); it != m_servicesVector.end(); ++it) {
      if ((*it)->getStartHandle() <= handle && handle <= (*it)->getEndHandle()) {
          pService = *it;
          break;
      }
    }

    if (pService != nullptr) {

	**pService->getCharacteristics(true);**		// added

        for (auto it = pService->begin(); it != pService->end(); ++it) {
            if ((*it)->getHandle() == handle) {
                return *it;
            }
        }
    }

    return nullptr;
}

h2zero, 2022-04-03:
Thank you for bringing this up. I guess it was assumed that the application would have retrieved the attribute database before using this function. Adding a check and retrieving the database if necessary is certainly a good idea in this case.

h2zero, 2022-04-03:
I have a question, what is your use case for this? I remember it was added when someone wanted to probe some devices seen on a scanner but I don't see much other use.

miksumin, 2022-04-04:
I am discovering communication with Mi Flora device without any documentation for it protocol just looking for someone else's code where they use handles. But as for now I found applications that makes a full discover of services/characteristics of my device.

h2zero, 2022-04-17:
I've decided this should not be changed. The application should retrieve the database with `discoverAttributes` or any other method before using this.

closed 2022-04-17 by h2zero