h2zero/NimBLE-Arduino #379: pBLEClient->setValue(ServiceUUID, CharacteristicUUID, value) not working

miksumin, 2022-04-03:
Hello, I found that pBLEClient->setValue(ServiceUUID, CharacteristicUUID, value) returns TRUE but not working properly.
The following code shows two ways of setting characteristic value:

uint8_t buf[2] = {0xA0, 0x1F};
NimBLEAttValue value(buf, 2);

// working code
BLERemoteCharacteristic* pRemoteCharacteristic = pBLEClient->getCharacteristic(0x0033);
bool result = pRemoteCharacteristic->writeValue(buf, 2, true);

// not working code
bool result = pBLEClient->setValue(ServiceUUID, CharacteristicUUID, value);

if (result)
	NimBLEAttValue value = pBLEClient->getValue(ServiceUUID, CharacteristicUUID);
	if (value.length() != 0) {
		const char *val = value.c_str();
		Serial.print("Received data (HEX): ");
		for (int i = 0; i < 16; i++) {
			Serial.print((uint8_t)val[i], HEX);
			Serial.print(" ");
		}
		Serial.println(" ");
	}
}

is there any ideas what could be a problem?

h2zero, 2022-04-03:
>// working code
BLERemoteCharacteristic* pRemoteCharacteristic = pBLEClient->getCharacteristic(0x0033);
bool result = pRemoteCharacteristic->writeValue(buf, 2, true);
>
>// not working code
bool result = pBLEClient->setValue(ServiceUUID, CharacteristicUUID, value);

Looks to me that the peer device does not allow writes without response. The "not working code" does not set the flag for response required, whereas the "working code" does.

Try this:  `pBLEClient->setValue(ServiceUUID, CharacteristicUUID, value, true);`

miksumin, 2022-04-04:
I tried it with response flag and it works! So you are right, my device does not allow writes without response. Many thanks!

h2zero, 2022-04-04:
Closing this as it's been resolved.

closed 2022-04-04 by h2zero