h2zero/NimBLE-Arduino #382: BLE Pairing Issue

Hassaan18, 2022-04-12:
The device asks for pairing every time I try to connect it to ESP

Hassaan18, 2022-04-12:
Here is the code for initialization

`bool ESP_BT::init()
{
    Serial.println("Starting NimBLE Server");

    /** set connection status as false */
    this->isConnected = false;
    this->commandInQueue = false;

    /** sets device name */
    NimBLEDevice::init((bt.bluetooth_name).c_str());

    /** Optional: set the transmit power, default is 3db */
    NimBLEDevice::setPower(ESP_PWR_LVL_P9); /** +9db */
    
    /** Set the IO capabilities of the device, each option will trigger a different pairing method.
     *  BLE_HS_IO_DISPLAY_ONLY    - Passkey pairing
     *  BLE_HS_IO_DISPLAY_YESNO   - Numeric comparison pairing
     *  BLE_HS_IO_NO_INPUT_OUTPUT - DEFAULT setting - just works pairing
     */
    //NimBLEDevice::setSecurityIOCap(BLE_HS_IO_DISPLAY_ONLY); // use passkey
    //NimBLEDevice::setSecurityIOCap(BLE_HS_IO_DISPLAY_YESNO); //use numeric comparison

    /** 2 different ways to set security - both calls achieve the same result.
     *  no bonding, no man in the middle protection, secure connections.
     *   
     *  These are the default values, only shown here for demonstration.   
     */ 
    //NimBLEDevice::setSecurityAuth(true, true, true); 
    NimBLEDevice::setSecurityAuth(BLE_SM_PAIR_AUTHREQ_BOND | BLE_SM_PAIR_AUTHREQ_MITM | BLE_SM_PAIR_AUTHREQ_SC);
    NimBLEDevice::setSecurityPasskey(123456);
    NimBLEDevice::setSecurityIOCap(BLE_HS_IO_DISPLAY_ONLY);

    pServer = NimBLEDevice::createServer();
    pServer->setCallbacks(new ServerCallbacks());

    NimBLEService* pDeadService = pServer->createService("DEAD");
    NimBLECharacteristic* pBeefCharacteristic = pDeadService->createCharacteristic(
                                               "BEEF",
                                               NIMBLE_PROPERTY::READ |
                                               NIMBLE_PROPERTY::WRITE |
                               /** Require a secure connection for read and write access */
                                               NIMBLE_PROPERTY::READ_ENC |  // only allow reading if paired / encrypted
                                               NIMBLE_PROPERTY::WRITE_ENC   // only allow writing if paired / encrypted
                                              );
  
    /**
     * TODO: change this
     */
    pBeefCharacteristic->setValue("Burger");
    pBeefCharacteristic->setCallbacks(&chrCallbacks);

    /** 2904 descriptors are a special case, when createDescriptor is called with
     *  0x2904 a NimBLE2904 class is created with the correct properties and sizes.
     *  However we must cast the returned reference to the correct type as the method
     *  only returns a pointer to the base NimBLEDescriptor class.
     */
    NimBLE2904* pBeef2904 = (NimBLE2904*)pBeefCharacteristic->createDescriptor("2904"); 
    pBeef2904->setFormat(NimBLE2904::FORMAT_UTF8);
    pBeef2904->setCallbacks(&dscCallbacks);
  

    NimBLEService* pBaadService = pServer->createService("BAAD");
    NimBLECharacteristic* pFoodCharacteristic = pBaadService->createCharacteristic(
                                               "F00D",
                                               NIMBLE_PROPERTY::READ |
                                               NIMBLE_PROPERTY::READ_ENC |
                                               //NIMBLE_PROPERTY::WRITE |
                                               NIMBLE_PROPERTY::NOTIFY
                                              );

    pFoodCharacteristic->setValue("Fries");
    pFoodCharacteristic->setCallbacks(&chrCallbacks);

    /** Note a 0x2902 descriptor MUST NOT be created as NimBLE will create one automatically
     *  if notification or indication properties are assigned to a characteristic.
     */

    /** Custom descriptor: Arguments are UUID, Properties, max length in bytes of the value */
    NimBLEDescriptor* pC01Ddsc = pFoodCharacteristic->createDescriptor(
                                               "C01D",
                                               NIMBLE_PROPERTY::READ, 
                                               //NIMBLE_PROPERTY::WRITE|
                                               //NIMBLE_PROPERTY::WRITE_ENC, // only allow writing if paired / encrypted
                                               20
                                              );
    pC01Ddsc->setValue("Send it back!");
    pC01Ddsc->setCallbacks(&dscCallbacks);

    /** Start the services when finished creating all Characteristics and Descriptors */  
    pDeadService->start();
    pBaadService->start();

    NimBLEAdvertising* pAdvertising = NimBLEDevice::getAdvertising();
    /** Add the services to the advertisment data **/
    pAdvertising->addServiceUUID(pDeadService->getUUID());
    pAdvertising->addServiceUUID(pBaadService->getUUID());
    /** If your device is battery powered you may consider setting scan response
     *  to false as it will extend battery life at the expense of less data sent.
     */
    pAdvertising->setScanResponse(true);
    pAdvertising->start();

    Serial.println("Advertising Started");
    return true;
}`

Hassaan18, 2022-04-13:
I guess, I have figured out the issue and found a workaround but that's not stable. The client address comes to be different every time the android device connects to the ESP. Is that a problem? For that, I have disabled the MITM but changing that alone didn't make any progress. But making the maximum bonds (CONFIG_BT_NIMBLE_MAX_BONDS) to 10 instead of 3 did improve the connectivity and the client address became steady as well. Why did the maximum bond number affect the client address? and it again doesn't work properly when I make it 8 or 20 or even 3

h2zero, 2022-04-13:
@Hassaan18 the address changing should not produce an issue if the bonding took place correctly as it should be a resolvable address. Could you set the log level to debug and provide the log output when the error occurs?

Hassaan18, 2022-04-13:
Yes, I have set the log level to debug. Do you want the logs of Android or of ESP?

Hassaan18, 2022-04-13:
> Yes, I have set the log level to debug. Do you want the logs of Android or of ESP?

Attached below are the debug logs
```
D NimBLEServer: ">> handleGapEvent: "
Client connected
Multi-connect support: start advertising
D NimBLEAdvertising: ">> Advertising start: customAdvData: 0, customScanResponseData: 0"
D NimBLEAdvertising: "<< Advertising start"
Client address: 4a:a1:c3:d1:a4:d3
D NimBLEServer: ">> handleGapEvent: "
D NimBLEServer: "Connection parameters updated."
D NimBLEServer: ">> handleGapEvent: "
D NimBLEServer: "Connection parameters updated."
D NimBLEServer: ">> handleGapEvent: "
D NimBLEServer: "Connection parameters updated."
D NimBLEServer: ">> handleGapEvent: "
Server Passkey Request
D NimBLEServer: "BLE_SM_IOACT_DISP; ble_sm_inject_io result: 0"
D NimBLEServer: "<< handleGATTServerEvent"
D NimBLEServer: ">> handleGapEvent: "
Starting BLE work!
D NimBLEServer: ">> handleGapEvent: "
D NimBLEServer: "<< handleGATTServerEvent"
D NimBLEServer: ">> handleGapEvent: "
I NimBLEServer: "subscribe event; attr_handle=8, subscribed: false"
D NimBLECharacteristicCallbacks: "onRead: default"
D NimBLECharacteristic: "Characteristic 0xf00d Read event"
D NimBLEServer: ">> handleGapEvent: "
I NimBLEServer: "subscribe event; attr_handle=8, subscribed: false"
D NimBLEServer: ">> handleGapEvent: "
Client disconnected - start advertising
D NimBLEAdvertising: ">> Advertising start: customAdvData: 0, customScanResponseData: 0"
W NimBLEAdvertising: "Advertising already active"
D NimBLEServerCallbacks: "onDisconnect(): Default"
D NimBLEAdvertising: ">> Advertising start: customAdvData: 0, customScanResponseData: 0"
W NimBLEAdvertising: "Advertising already active"
D NimBLEServer: ">> handleGapEvent: "
Client connected
Multi-connect support: start advertising
D NimBLEAdvertising: ">> Advertising start: customAdvData: 0, customScanResponseData: 0"
D NimBLEAdvertising: "<< Advertising start"
Client address: 7b:9a:a8:90:99:81
D NimBLEServer: ">> handleGapEvent: "
D NimBLEServer: "Connection parameters updated."
D NimBLEServer: ">> handleGapEvent: "
D NimBLEServer: "Connection parameters updated."
D NimBLEServer: ">> handleGapEvent: "
D NimBLEServer: "Connection parameters updated."
D NimBLEServer: ">> handleGapEvent: "
Server Passkey Request
D NimBLEServer: "BLE_SM_IOACT_DISP; ble_sm_inject_io result: 0"
D NimBLEServer: "<< handleGATTServerEvent"
D NimBLEServer: ">> handleGapEvent: "
D NimBLEServer: ">> disconnect()"
D NimBLEServer: "<< disconnect()"
Encrypt connection failed - disconnecting client
D NimBLEServer: ">> handleGapEvent: "
Client disconnected - start advertising
D NimBLEAdvertising: ">> Advertising start: customAdvData: 0, customScanResponseData: 0"
W NimBLEAdvertising: "Advertising already active"
D NimBLEServerCallbacks: "onDisconnect(): Default"
D NimBLEAdvertising: ">> Advertising start: customAdvData: 0, customScanResponseData: 0"
W NimBLEAdvertising: "Advertising already active"
```

Hassaan18, 2022-04-13:
> Yes, I have set the log level to debug. 

Please find below the logs for ESP32

```
Starting BLE work!
Starting NimBLE Server
I NimBLEDevice: "BLE Host Task Started"
I NimBLEDevice: "NimBle host synced."
D NimBLEDevice: ">> setPower: 7 (type: 11)"
D NimBLEDevice: "<< setPower"
D NimBLEDevice: "Setting bonding: 1, mitm: 1, sc: 1"
D NimBLEServer: ">> createService - 0xdead"
D NimBLEServer: "<< createService"
D NimBLECharacteristic: ">> setValue: length=7, data=42757267657200, characteristic UUID=0xbeef"
D NimBLECharacteristic: "<< setValue"
D NimBLEUUID: "Comparing UUIDs; type 16 to 16; UUID 0x2904 to 0x2902"
D NimBLEUUID: "Comparing UUIDs; type 16 to 16; UUID 0x2904 to 0x2904"
D NimBLEServer: ">> createService - 0xbaad"
D NimBLEUUID: "Comparing UUIDs; type 16 to 16; UUID 0xdead to 0xbaad"
D NimBLEServer: "<< createService"
D NimBLECharacteristic: ">> setValue: length=6, data=467269657300, characteristic UUID=0xf00d"
D NimBLECharacteristic: "<< setValue"
D NimBLEUUID: "Comparing UUIDs; type 16 to 16; UUID 0xc01d to 0x2902"
D NimBLEUUID: "Comparing UUIDs; type 16 to 16; UUID 0xc01d to 0x2904"
D NimBLEService: ">> start(): Starting service: UUID: 0xdead, handle: 0xffff"
D NimBLEService: "Adding 1 characteristics for service UUID: 0xdead, handle: 0xffff"
D NimBLEService: "<< start()"
D NimBLEService: ">> start(): Starting service: UUID: 0xbaad, handle: 0xffff"
D NimBLEService: "Adding 1 characteristics for service UUID: 0xbaad, handle: 0xffff"
D NimBLEService: "<< start()"
D NimBLEAdvertising: ">> Advertising start: customAdvData: 0, customScanResponseData: 0"
primary service
           uuid 0x1800
         handle 1
     end_handle 5
characteristic
           uuid 0x2a00
     def_handle 2
     val_handle 3
   min_key_size 0
          flags [READ]
characteristic
           uuid 0x2a01
     def_handle 4
     val_handle 5
   min_key_size 0
          flags [READ]
primary service
           uuid 0x1801
         handle 6
     end_handle 9
characteristic
           uuid 0x2a05
     def_handle 7
     val_handle 8
   min_key_size 0
          flags [INDICATE]
ccc descriptor
           uuid 0x2902
         handle 9
   min_key_size 0
          flags [READ|WRITE]
primary service
           uuid 0xdead
         handle 10
     end_handle 13
characteristic
           uuid 0xbeef
     def_handle 11
[... 232 lines cut ...]
- Service Changed [I] (0x2A05)
   Client Characteristic Configuration (0x2902)
Unknown Service (0000dead-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [R W] (0000beef-0000-1000-8000-00805f9b34fb)
   Characteristic Presentation Format (0x2904)
Unknown Service (0000baad-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [N R] (0000f00d-0000-1000-8000-00805f9b34fb)
   Client Characteristic Configuration (0x2902)
   Unknown Descriptor (0000c01d-0000-1000-8000-00805f9b34fb)
D	02:14:37.188	gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
D	02:14:37.189	gatt.setCharacteristicNotification(0000f00d-0000-1000-8000-00805f9b34fb, true)
I	02:14:37.264	Connection parameters updated (interval: 45.0ms, latency: 0, timeout: 600ms)
D	02:14:44.805	[Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: BOND_BONDED (12)
I	02:14:44.805	Device bonded
V	02:14:46.398	Reading characteristic 0000f00d-0000-1000-8000-00805f9b34fb
D	02:14:46.398	gatt.readCharacteristic(0000f00d-0000-1000-8000-00805f9b34fb)
I	02:14:46.625	Read Response received from 0000f00d-0000-1000-8000-00805f9b34fb, value: (0x) 25-53-25-31-35-2C-61-62-63-2D-31-32-33-2D-74-69-6D-65-3D-35-35-25-53-25-24-00
A	02:14:46.625	"(0x) 25-53-25-31-35-2C-61-62-63-2D-31-32-33-2D-74-69-6D-65-3D-35-35-25-53-25-24-00" received
D	02:14:48.792	[Server callback] Connection state changed with status: 0 and new state: DISCONNECTED (0)
I	02:14:48.792	[Server] Device disconnected
D	02:14:48.792	[Callback] Connection state changed with status: 8 and new state: DISCONNECTED (0)
E	02:14:48.800	Error 8 (0x8): GATT CONN TIMEOUT
I	02:14:48.800	Disconnected
D	02:14:48.821	[Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED
```

h2zero, 2022-04-13:
Thanks, I believe this is a bug upstream. If you erase_flash before uploading does it work?

Hassaan18, 2022-04-13:
It still gets the same response

Hassaan18, 2022-04-13:
I wonder why didn't I think that way. Thanks a lot, I tried erasing the flash and now it's working fine. Let me just check it on other devices as well

Hassaan18, 2022-04-14:
> I wonder why didn't I think that way. Thanks a lot, I tried erasing the flash and now it's working fine. Let me just check it on other devices as well

@h2zero I have tried using it on other devices and ESPs as well and it works fine on v1.2.0 but when I change the version to 1.3 Series, the ESP is getting Panic'ed when I click on Pairing option on my android device

h2zero, 2022-04-14:
Could you elaborate a bit? What do you mean by panic when using FreeRTOS? Can you share a log of this?

Hassaan18, 2022-04-14:
> Could you elaborate a bit? What do you mean by panic when using FreeRTOS? Can you share a log of this?

The code works totally fine unless I try to connect with the ESP on Bluetooth. Once I connect to the device, nrf asks for pairing and when I click on that, the core Panics.

The tasks are mentioned below

```

    xTaskCreatePinnedToCore(vStatusLed, "Status LED", 5000, NULL, 1, &ledTask, 1);
    xTaskCreatePinnedToCore(vAcquireData, "Data Acquisition", 5000, NULL, 3, &dataTask1, 1);
    xTaskCreatePinnedToCore(vStorage, "Storage Handler", 5000, NULL, 2, &storageTask, 1);
    xTaskCreatePinnedToCore(vBlTransfer, "Bluetooth Transfer", 4000, NULL, 3, &blTask2, 0);
    xTaskCreatePinnedToCore(vWifiTransfer, "Transfer data on Wifi", 7000, NULL, 1, &wifiTask, 0);
```

Here are the logs for ESP32

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8

Starting NimBLE Server
I NimBLEDevice: "BLE Host Task Started"
I NimBLEDevice: "NimBle host synced."
D NimBLEDevice: ">> setPower: 7 (type: 11)"
D NimBLEDevice: "<< setPower"
D NimBLEDevice: "Setting bonding: 1, mitm: 1, sc: 1"
D NimBLEServer: ">> createService - 0xdead"
D NimBLEServer: "<< createService"
D NimBLECharacteristic: ">> setValue: length=7, data=42757267657200, characteristic UUID=0xbeef"
D NimBLECharacteristic: "<< setValue"
D NimBLEServer: ">> createService - 0xbaad"
D NimBLEServer: "<< createService"
D NimBLECharacteristic: ">> setValue: length=6, data=467269657300, characteristic UUID=0xf00d"
D NimBLECharacteristic: "<< setValue"
D NimBLEService: ">> start(): Starting service: UUID: 0xdead, handle: 0xffff"
D NimBLEService: "Adding 1 characteristics for service UUID: 0xdead, handle: 0xffff"
D NimBLEService: "<< start()"
D NimBLEService: ">> start(): Starting service: UUID: 0xbaad, handle: 0xffff"
D NimBLEService: "Adding 1 characteristics for service UUID: 0xbaad, handle: 0xffff"
D NimBLEService: "<< start()"
D NimBLEAdvertising: ">> Advertising start: customAdvData: 0, customScanResponseData: 0"
primary service
           uuid 0x1800
         handle 1
     end_handle 5
characteristic
           uuid 0x2a00
     def_handle 2
     val_handle 3
   min_key_size 0
          flags [READ]
characteristic
           uuid 0x2a01
     def_handle 4
     val_handle 5
   min_key_size 0
          flags [READ]
primary service
           uuid 0x1801
         handle 6
     end_handle 9
characteristic
[... 163 lines cut ...]
V	18:02:51.777	Generic Access (0x1800)
- Device Name [R] (0x2A00)
- Appearance [R] (0x2A01)
Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
   Client Characteristic Configuration (0x2902)
Unknown Service (0000dead-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [R W] (0000beef-0000-1000-8000-00805f9b34fb)
   Characteristic Presentation Format (0x2904)
Unknown Service (0000baad-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [N R] (0000f00d-0000-1000-8000-00805f9b34fb)
   Client Characteristic Configuration (0x2902)
   Unknown Descriptor (0000c01d-0000-1000-8000-00805f9b34fb)
D	18:02:51.777	gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
D	18:02:51.779	gatt.setCharacteristicNotification(0000f00d-0000-1000-8000-00805f9b34fb, true)
I	18:02:51.837	Connection parameters updated (interval: 45.0ms, latency: 0, timeout: 2400ms)
D	18:02:54.999	[Server callback] Connection state changed with status: 0 and new state: DISCONNECTED (0)
I	18:02:54.999	[Server] Device disconnected
D	18:02:55.010	[Callback] Connection state changed with status: 8 and new state: DISCONNECTED (0)
E	18:02:55.011	Error 8 (0x8): GATT CONN TIMEOUT
I	18:02:55.011	Disconnected
D	18:02:55.070	[Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: BOND_NONE (10)
I	18:02:55.070	Bonding failed
D	18:02:55.132	[Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED
```

h2zero, 2022-04-14:
Ok, I see now, thanks! If you decode that backtrace I believe you will find the cause of this to be due to accessing a null pointer, you can see this in the backtrace `EXCVADDR: 0x00000000`. Please try to decode the backtrace and post it here.

Hassaan18, 2022-04-14:
Ok, I'll decode the backtrace and get back to you

Hassaan18, 2022-04-14:
The issue seems to be resolved for now. I have multiple versions of library installed in libdeps in .pio, hence causing issues in stack. Thankyou so much for the help

Hassaan18, 2022-04-15:
The issue was regarding the contradiction in the stack since both library versions are stored in libdeps folder. I removed version 1.2.0 and it's working fine now. But I'm not able to use NimBLEDevice::getNumBonds() function. It's causing the core to panic with backtrace

```
[I][bluetooth.cpp:181] init(): Starting NimBLE Server
assertion "mu->handle" failed: file ".pio\libdeps\esp32doit-devkit-v1\NimBLE-Arduino\src\porting\npl\freertos\src\npl_os_freertos.c", line 185, function: npl_freertos_mutex_pend
abort() was called at PC 0x4012424b on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4009001c:0x3ffcd800 0x40090299:0x3ffcd820 0x4012424b:0x3ffcd840 0x400eb8bf:0x3ffcd870 0x400e3db4:0x3ffcd890 0x400e3dcb:0x3ffcd8b0 0x400e9a63:0x3ffcd8d0 0x400e9c45:0x3ffcd8f0 0x400e9cff:0x3ffcd9a0 0x400dbb9a:0x3ffcd9d0 0x400d5489:0x3ffcda10 0x400d8d47:0x3ffcda50 0x401085c6:0x3ffcdae0 0x400912aa:0x3ffcdb00
  #0  0x4009001c:0x3ffcd800 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #1  0x40090299:0x3ffcd820 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #2  0x4012424b:0x3ffcd840 in __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)   
  #3  0x400eb8bf:0x3ffcd870 in npl_freertos_mutex_pend at .pio\libdeps\esp32doit-devkit-v1\NimBLE-Arduino\src\porting\npl\freertos\src/npl_os_freertos.c:371
  #4  0x400e3db4:0x3ffcd890 in ble_npl_mutex_pend at .pio\libdeps\esp32doit-devkit-v1\NimBLE-Arduino\src/nimble/nimble_npl_os.h:109
      (inlined by) ble_hs_lock_nested at .pio\libdeps\esp32doit-devkit-v1\NimBLE-Arduino\src\nimble\host\src/ble_hs.c:165
  #5  0x400e3dcb:0x3ffcd8b0 in ble_hs_lock at .pio\libdeps\esp32doit-devkit-v1\NimBLE-Arduino\src/nimble/nimble_npl_os.h:109
  #6  0x400e9a63:0x3ffcd8d0 in ble_store_read at .pio\libdeps\esp32doit-devkit-v1\NimBLE-Arduino\src\nimble\host\src/ble_store.c:321
  #7  0x400e9c45:0x3ffcd8f0 in ble_store_iterate at .pio\libdeps\esp32doit-devkit-v1\NimBLE-Arduino\src\nimble\host\src/ble_store.c:360
  #8  0x400e9cff:0x3ffcd9a0 in ble_store_util_bonded_peers at .pio\libdeps\esp32doit-devkit-v1\NimBLE-Arduino\src\nimble\host\src/ble_store_util.c:239 (discriminator 1)
  #9  0x400dbb9a:0x3ffcd9d0 in NimBLEDevice::getNumBonds() at .pio\libdeps\esp32doit-devkit-v1\NimBLE-Arduino\src/NimBLEDevice.cpp:855 (discriminator 1)
  #10 0x400d5489:0x3ffcda10 in ESP_BT::init() at src/bluetooth.cpp:399
  #11 0x400d8d47:0x3ffcda50 in setup() at src/main.cpp:212
  #12 0x401085c6:0x3ffcdae0 in loopTask(void*) at C:\Users\hassa\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:18
  #13 0x400912aa:0x3ffcdb00 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

```

h2zero, 2022-04-15:
I believe the issue here is calling those functions before initializing the stack.

Hassaan18, 2022-04-15:
How and where should I initialize the stack?

h2zero, 2022-04-15:
That's what happens when you call `NimBLEDevice::init`, make sure you call this before making any other calls into NimBLE.

Hassaan18, 2022-04-15:
Yeah, that was the issue. Thankyou so much for the assistance

Hassaan18, 2022-04-15:
One last thing, why have you limited the Maximum Bond number to 3 and Maximum length to 512? Can you increase this in the library?

h2zero, 2022-04-15:
>why have you limited the Maximum Bond number to 3

It's the default value in the espressif config. A larger value consumes more resources and not many users would need more than this.

>Maximum length to 512

That is specified the the Bluetooth specification.

Hassaan18, 2022-04-15:
Can't we increase the length in sdkconfig.h?

h2zero, 2022-04-15:
No, it would not work if you could. You would violate the BLE specification, which means other devices would not work with it.

Hassaan18, 2022-04-15:
Is there any possible thing then? I want to transmit a string of around 600 Bytes every second to my BLE-based Android Application. Doesn't it work with the usual Bluetooth when we transfer files over that?

h2zero, 2022-04-15:
Sure, you'll just need to get creative. Packetize the data and send multiple notifications, recombine on the phone. Or some other method you can think of.

Longer data was only available with classic Bluetooth.

Hassaan18, 2022-04-15:
Yeah, packetizing the data is the last option. Trying if I could handle it on ESP side so that works with other apps like NRF as well. Let's see if any other solution seem relevant. And Thankyou again. That was all

Hassaan18, 2022-04-16:
> Yeah, packetizing the data is the last option. Trying if I could handle it on ESP side so that works with other apps like NRF as well. Let's see if any other solution seem relevant.

@h2zero I have tried increasing the parameter BLE_ATT_ATTR_MAX_LEN in ble_att.h and it works perfectly now. How's about updating this parameter. Does it have some bad effects?

h2zero, 2022-04-16:
Lol, that's one way to do it. I suspect you may find some devices will not work with this configuration, as it's not spec compliant, but whatever works for you.

Hassaan18, 2022-04-16:
That would happen on older versions of Bluetooth i.e., less than Bluetooth 4.2. Right?

Hassaan18, 2022-04-16:
> Lol, that's one way to do it. I suspect you may find some devices will not work with this configuration, as it's not spec compliant, but whatever works for you.

This would happen only for Bluetooth versions lesser than 5.0. Right?

h2zero, 2022-04-16:
The max characteristic size has not changed with BT5, that's why it's still hard coded that way in the stack.

Hassaan18, 2022-04-16:
Ok, I see. That was all. Thank you again for your time. I appreciate your efforts towards this library.

Hassaan18, 2022-04-19:
> Sure, you'll just need to get creative. Packetize the data and send multiple notifications, recombine on the phone. Or some other method you can think of.
@h2zero can you please tell what should be the minimum delay between 2 packets?

h2zero, 2022-04-20:
I would recommend using the onStatus callback to know when it's safe to send the next data. To answer your question, you should delay for the connection interval time, which you can get from `NimBLEServer::getPeerInfo.getConnInterval`.

Hassaan18, 2022-04-20:
How to use the onStatus callback since it always returns the status code as 1. Shouldn't it be onRead callback?

I couldn't find the getConnInterval function within NimBLEServer class. Can you please specify where it is?

Hassaan18, 2022-04-20:
> How to use the onStatus callback since it always returns the status code as 1. Shouldn't it be onRead callback? I am not using the notifications to read the data frame, but a fixed sampling rate within the Android App.
> 
> Let me try waiting for the connection interval if it works.

`getCOnnInterval` gives me 36 and putting that delay, even higher delays like 1000ms is causing both the strings to mix up and data packets received are not correct.

h2zero, 2022-04-22:
I assumed you were using notifications. If the app is reading from the device you could just swap the data after the read callback is called then read it again, no delay required.

Hassaan18, 2022-04-24:
> I assumed you were using notifications. If the app is reading from the device you could just swap the data after the read callback is called then read it again, no delay required.

the issue is that the data is not being received correctly. both portions are being mixed up on the receiving end

h2zero, 2022-04-25:
There isn't anything happening in the library that would affect this. The issue I think you're having is the MTU size is smaller than your packets so a long read is being performed by the client. It may be faster for you to use a second characteristic to split the data and the client can just read both and combine the data.

Hassaan18, 2022-04-25:
But when I send data packets with size less than 512 continuously (let's say after every 1s), that makes no issue in that. But when I send 2 segments of a data frame (one of 400 and other of 300 bytes), it start having the issues

Hassaan18, 2022-04-27:
@h2zero Are the Service names predefined in the BLE specifications? Since, when I create a service named MEAT, it creates a service named 0x0000 while the Characteristic name is fine (0xdeaf).

h2zero, 2022-04-27:
UUIDs are hexadecimal so your string value is invalid and sets the uuid to 0000. The library just accepts a string for convenience.

Hassaan18, 2022-04-27:
> UUIDs are hexadecimal so your string value is invalid and sets the UUID to 0000. The library just accepts a string for convenience.

Oh, got it. Thanks!

Hassaan18, 2022-04-27:
> But when I send data packets with sizes less than 512 continuously (let's say every 500ms), that makes no issue in that. But when I send 2 segments of a data frame (one of 400 and the other of 300 bytes), it starts having the issues

And what's about this one?

h2zero, 2022-04-28:
I don't know what is happening in that case as it would appear to be a user implementation problem. 

I would need to see an example of the code and logs to diagnose.

closed 2022-06-20 by Hassaan18