h2zero/NimBLE-Arduino #385: How to make client response it's Device name ？

somaker, 2022-04-16:
First of all, thank you very much, this is a very good and helpful project!

I use eps32 C3 as a client to connect to a device. When pairing, the name of the client will be displayed on the device, but this process has been unable to complete, and the pairing cannot be completed. 
Through wireshark, I see an error when the device requests the client's Device Name. 

691	1.921	Slave_0x25c1c4c7	LE 1M	ATT	11	152µs	0	0	False	24	Rcvd Read By Type Request, Device Name, Handles: 0x0001..0xffff

692	1.938	Master_0x25c1c4c7	LE 1M	ATT	9	24599µs	0	1	False	25	Sent Error Response - Attribute Not Found, Handle: 0x0001 (Unknown)
 
 **Details of the first record：**
Bluetooth Attribute Protocol
    Opcode: Read By Type Request (0x08)
    Starting Handle: 0x0001
    Ending Handle: 0xffff
    UUID: Device Name (0x2a00)

 **Details of the second record：**
Bluetooth Attribute Protocol
    Opcode: Error Response (0x01)
    Request Opcode in Error: Read By Type Request (0x08)
    Handle in Error: 0x0001 (Unknown)
    Error Code: Attribute Not Found (0x0a)
    [UUID: Device Name (0x2a00)]

I have set the client name by NimBLEDevice::init("YT12345");

**What should I do to make the client return its own name correctly?** 

Thank you！

I use h2zero/NimBLE-Arduino @ ^1.3.6

h2zero, 2022-04-16:
Interesting use case, can't say I've ever tried to retrieve the clients device name. I would try creating a server instance and just calling it's start method.

somaker, 2022-04-16:
I also tested the original Arduino BLE Library and it works.

I did the following configuration

BLEDevice::init("YT123456");
BLEDevice::setEncryptionLevel(ESP_BLE_SEC_ENCRYPT);

When pairing, the slave device gets the device name: "ESP32", and pairing can be completed.
Although the name was not set by me, the slave got the device name

h2zero, 2022-04-16:
Interesting, I'll try this when I get some time. The NimBLE stack does seem to handle this differently from bluedroid. 

If you add this after initialization does it work?
```
NimBLEDevice::createServer()->start();
```

somaker, 2022-04-20:
> Interesting, I'll try this when I get some time. The NimBLE stack does seem to handle this differently from bluedroid.
> 
> If you add this after initialization does it work?
> 
> ```
> NimBLEDevice::createServer()->start();
> ```

Yes，It‘s works！Thank you very much！

closed 2022-04-20 by somaker