h2zero/NimBLE-Arduino #386: How Slave Gets Encryption Error Information

somaker, 2022-04-20:
I created a HID Server and paired it with the phone. 

When I use 

NimBLEDevice::deleteAllBonds();  

to delete the binding information, the phone can still connect to the server,  but it will be disconnected immediately.
The ondisconnect event triggers, but I don't know how to get the reason for the disconnection.
Through sniffer, we know that the reason  is "PIN or Key Missing"
We can't trigger a re-pair unless we choose to ignore this hid device on the phone.
My question is, is it possible to get the error message and even request re-pairing from the slave side?

 BLE Sniffer:

Bluetooth Low Energy Link Layer
    Access Address: 0xaf9aabed
    [Master Address: 67:86:ee:88:51:3a (67:86:ee:88:51:3a)]
    [Slave Address: Espressi_56:db:7a (a0:76:4e:56:db:7a)]
    Data Header: 0x0317
    Control Opcode: LL_REJECT_IND_EXT (0x11)
    Reject Opcode: LL_ENC_REQ (0x03)
    Error Code: PIN or Key Missing (0x06)
    CRC: 0x8e0ec0

Thank you!

h2zero, 2022-04-23:
If the error only comes after the disconnection then it would be too late to trigger the pairing anyway. The client needs to delete the bond on it's side as well. I will look to change the callback in the future to provide the reason code.

h2zero, 2024-06-04:
Reason code is now provided in the onDisconnect callback.

closed 2024-06-04 by h2zero