h2zero/NimBLE-Arduino #393: Version number in library.properties does not match release versions

Bascy, 2022-04-30:
Release 1.3.7 and 1.3.8 both contain version nr 1.3.6 in library.properties.
This can cause problems when using a dependency on those specific versions

i.e.
  dependency on https://github.com/h2zero/NimBLE-Arduino @ ^1.3.7 will raise an error as the downloaded lib with tag 1.3.7 contains 1.3.6 version label this will cause the following error:

> Error: Package version 1.3.6+sha.8b5cfb0 doesn't satisfy requirements ^1.3.7 based on PackageMetaData <type=library name=NimBLE-Arduino version=1.3.6+sha.8b5cfb0 spec={'owner': None, 'id': None, 'name': 'NimBLE-Arduino', 'requirements': '^1.3.7', 'uri': 'git+https://github.com/h2zero/NimBLE-Arduino'}

Bascy, 2022-04-30:
See PR #394

h2zero, 2022-04-30:
This is due to using a release branch for 1.3.X versions, if you use `h2zero/NimBLE-Arduino@^1.3.7` instead of pointing to the git repo it works fine. The master branch isn't part of any release yet so the version hasn't been updated. 

Thanks for the PR I will merge it to resolve this.

closed 2022-05-13 by Bascy