h2zero/NimBLE-Arduino #402: Crash when stoping a passive and starting an active scan.

wesee-dev, 2022-05-14:
The ESP32 crashes most of the time when an active scan is requested after canceling a passive scan.
I tried to add a delay before requesting the scan but no luck.

Logs:
'''
D NimBLEScan: ">> stop()"
D NimBLEScan: "<< stop()"
[Info] Starting scan...
D NimBLEScan: ">> start(duration=30)"

assert failed: host_rcv_pkt C:\C3-5i\C3_5i\libraries\NimBLE-Arduino-1.3.8\src\esp-hci\src\esp_nimble_hci.c:418 (0)

Backtrace:0x40084d75:0x3ffbea4c |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting... 
'''

h2zero, 2022-05-14:
Hello, the assert that triggered is a controller failure. I've attempted to reproduce this with the following code and was not successful.

```
#include "NimBLEDevice.h"
int scanTime = 5; //In seconds
BLEScan* pBLEScan;

bool active = false;

class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
  /*** Only a reference to the advertised device is passed now
    void onResult(BLEAdvertisedDevice advertisedDevice) { **/
    void onResult(BLEAdvertisedDevice* advertisedDevice) {
  /** Serial.printf("Advertised Device: %s \n", advertisedDevice.toString().c_str()); **/
      Serial.printf("Advertised Device: %s \n", advertisedDevice->toString().c_str());
    }
};

/** Callback to process the results of the last scan or restart it */
void scanEndedCB(NimBLEScanResults results){
    Serial.println("Scan Ended");
    active = !active;
}

void setup() {
  Serial.begin(115200);
  Serial.println("Scanning...");

  BLEDevice::init("");
  pBLEScan = BLEDevice::getScan(); //create new scan
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(active); //active scan uses more power, but get results faster
  pBLEScan->setInterval(100);
  pBLEScan->setWindow(99);  // less or equal setInterval value
}

void loop() {
    if (!pBLEScan->isScanning()) {
        pBLEScan->setActiveScan(active);
        Serial.printf("scan start, active = %u\n", active);
        pBLEScan->start(scanTime, scanEndedCB, false);
    }
  delay(10);
}
```

To further diagnose I would need more code to to reproduce the issue.

wesee-dev, 2022-05-14:
Hello, thanks for the quick repky.
I also could not reproduce the issue using the provided code, even if the device was already scanning I had no issues in switching the scan modes.
But I solved the issue by requesting another task to trigger the scan using a queue. Which is probably the right thing to do in my project.
The scan request was previously being issued directly from a callback of ESPAsyncWebServer. So this is probably the cause.
But what bugs me is that if the scan type is not changed, then no crashes occur.

closed 2022-05-16 by wesee-dev