h2zero/NimBLE-Arduino #403: getServiceUUIDCount for custom service UUID

alberk8, 2022-05-15:
Hi,

When I use the following code, I get a return of 2 and these are Bluetooth SIG compliant UUID,  How do I check custom UUID in AdvertisedDeviceCallbacks ? 

For example the following code will always be false.
`advertisedDevice->isAdvertisingService(NimBLEUUID("00001523-1212-efde-1523-785feabcd123")`

`Serial.println(advertisedDevice->getServiceDataCount());`

h2zero, 2022-05-15:
Do you know the UUID's it's advertising? If that returns false the the UUID was not in the advertisement data. 

Try printing the ones that are there:
```
uint8_t count =  advertisedDevice->getServiceUUIDCount();
for (uint8_t i = 0; i < count ; i++) {
  Serial.printf("UUID %u = %s\n", i,  advertisedDevice->getServiceUUID(i).toString().c_str());
}
````

alberk8, 2022-05-16:
thank you for the prompt reply.   Checking with the code you have provided above the custom UUID was not included in the advertisement data.  

One more question,  Is there any documentation as to the preference of notify vs indicate if both are provided by the Server/Peripheral device?

h2zero, 2022-05-16:
Indications are only potentially useful for the server (the server application knows the data was received), but this causes delays and really doesn't offer much benefit. If notifications are available just use those.

closed 2022-07-09 by h2zero