h2zero/NimBLE-Arduino #406: How to set address type RANDOM

paulhamsh, 2022-05-21:
Hi
I have a device I am trying to emulate and in Wireshark scan response and advertising response it shows as having TxAdd: RANDOM with address 0xdb, 0xab, 0x7f, 0xe0, 0x2f,  0x5c.
How can I change the settings on the ESP32 to achieve that? I don't need a random address, I just want it to show as RANDOM, not as PUBLIC.

I think it can only be done with esp calls, but the code below seems not to work - either the call to esp_ble_gap_set_rand_addr or esp_ble_gap_config_local_privacy
The esp_base_mac_addr_set doesn't fail, the other two do.
But in all cases it stops Lightblue from scanning the ESP32.
I'm not sure where in the code to put the esp calls, either.

    BLEDevice::init("XXXXXXXX"); // 
    BLEDevice::setPower(ESP_PWR_LVL_P9); /** +9db */

    esp_bd_addr_t addr = {0xdb, 0xab, 0x7f, 0xe0, 0x2f,  0x5c}; 
//    if (esp_ble_gap_set_rand_addr(addr) != ESP_OK) {  Heltec.display->clear(); Heltec.display->drawString(0, 30, "FAIL RAND"); Heltec.display->display();};
//    if (esp_ble_gap_config_local_privacy(true) != ESP_OK) {  Heltec.display->clear(); Heltec.display->drawString(0, 30, "FAIL PRIV"); Heltec.display->display();};
    if (esp_base_mac_addr_set(addr) != ESP_OK) {  Heltec.display->clear(); Heltec.display->drawString(0, 30, "FAIL MAC"); Heltec.display->display();};
//    esp_ble_gap_clear_rand_addr();
    
    pSCServer = BLEDevice::createServer();
    pSCServer->setCallbacks(new SCServerCallbacks());

h2zero, 2022-05-21:
There is no API to do this but I believe if you modify the library slightly it may work.

change https://github.com/h2zero/NimBLE-Arduino/blob/8ff8f0fc35afed5db8276bd2f41afce22732e1f7/src/NimBLEDevice.cpp#L87
to BLE_OWN_ADDR_RANDOM

paulhamsh, 2022-05-21:
When I do that it doesn't show in LightBlue scan - just vanishes. Should I change the address as well to meet the 'random' spec?

paulhamsh, 2022-05-21:
What is really weird is if I make that change, load it, then undo the change, and re-load it - it works with the app that I needed it to work with, Without making the change it needs LightBlue (or other) to scan / connect first time then it is fine, with this change it doesn't need that. And it doesn't seem to be a ios caching thing because I just erased the iphone and it works fine. 
So it seems to make a difference to the ESP to have, at some point, been run with the RANDOM setting.
Could that be true?

paulhamsh, 2022-05-21:
Just tried that again on an M5Stack and it doesn't happen on that - so perhaps just a fluke

h2zero, 2022-05-21:
Interesting, unfortunately the way the esp32 handles the address and address type makes it somewhat difficult due to needing host based privacy. In addition to that the functions needed in NimBLE to set this aren't available without some code hacks to the core files. 

If you're adventurous you can walk through the code call stack of `NimBLEDevice::setOwnAddrType` to find the required calls to set the address and type there.

paulhamsh, 2022-05-22:
It looks like it works if you change the NimBLEDevice.cpp as you suggested, and also add the following, just before advertising:

```
    ble_addr_t blead;
    int rc;

    rc = ble_hs_id_gen_rnd(1, &blead);
    if (rc != 0) Serial.println("Rand failed");

    rc = ble_hs_id_set_rnd(blead.val);
    if (rc != 0) Serial.println("Addr failed");

    pAdvertising->setScanResponse(true);
    pAdvertising->start();
```

h2zero, 2022-05-22:
Ha, well done! I thought you were trying to set a specific address not just a random address of type random?

Awesome that this worked :smile:

paulhamsh, 2022-05-22:
I thought I needed to (or, at least, that that would be easier) but this worked fine.

h2zero, 2024-06-04:
This functionality has been added in the master branch.

closed 2024-06-04 by h2zero

paulhamsh, 2024-08-06:
Hi - can you tell me where it was added? I'd love to see what the new code is.

h2zero, 2024-08-08:
@paulhamsh, the function is here https://github.com/h2zero/NimBLE-Arduino/blob/b6a169f1f2e939ce0ca11d003cc38ef141698d6a/src/NimBLEDevice.cpp#L1106