h2zero/NimBLE-Arduino #407: Is it possible to add a timeout to report scan result if BLE_HCI_ADV_RPT_EVTYPE_SCAN_RSP is not received

wesee-dev, 2022-05-25:
Hello,
I have a few Xiaomi and Qingping devices that advertise packets when an event occurs, like movement detected or door/window open and close.
If the scan type is active, this devices send a scan response of type BLE_HCI_ADV_TYPE_ADV_IND,containing the data. 
However since the scan is active the scan result will only be reported after BLE_HCI_ADV_RPT_EVTYPE_SCAN_RSP event is received. So the sensor data/event is lost, since this devices don't always send a full response (maybe to save battery).
Of course I can solve the issue by using a passive scan, but in that way I can't use them together with sensors that requires an active scan.
Is it possible through the NimBLE stack, or by implementing directly in NimScan.cpp to add a timeout, that would cause the response to be reported if the sensor/device doesn't send the full scan result (BLE_HCI_ADV_RPT_EVTYPE_SCAN_RSP) after some time.
Thank

h2zero, 2022-05-25:
This has been on my mind for a while as well but I could not come up with an appropriate action. If we use a timeout what should it be?, as it could come immediately after or never. 

Unfortunately there is no general solution to this problem as the stack does not provide a timeout mechanism either.

The only solution that would make much sense is to provide a callback for just the advertisement data and then the full data or just scan response after.

wesee-dev, 2022-05-25:
This is a hard issue indeed.
Maybe this can be solved by previously setting/registering the devices, via its BD-ADDR, that should not wait for a full response before being reported. Kind of like a whitelist mechanism. 
I think registering different callbacks can easily lead to code duplication on the user side. But it's a valid solution and it would work just fine. 
Thanks for the quick response :)

h2zero, 2022-05-25:
The way this works now does give an option though. If you set the scan time to something short, say 3 seconds, the callback will be called for the device regardless of the scan response status. From there the scan ended callback will be called and you can restart it from there.

wesee-dev, 2022-05-25:
That's nice to know. I think it will fit in my project.
Thanks for the help =D

closed 2022-05-25 by wesee-dev