h2zero/NimBLE-Arduino #409: Fatal Include Error

trevorsb, 2022-05-31:
Hi,

I use NimBLE-Arduino Library with ESP32-WROOM-32E. I port it to my chip through Platform.io. It has worked very well up until today. Today, when I try to compile my code, I get an include error `.pio/libdeps/esp32dev/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs_hci_cmd.c:31:26: fatal error: soc/soc_caps.h: No such file or directory
compilation terminated.` 

I've tried to compile several of my branches that were compiling without error just yesterday, and I get this error every time. The only thing I can think of is some sort of versioning issue, where something has been updated or gone out of date. Any ideas would be appreciated. Thank you
[image]
!

h2zero, 2022-05-31:
Looks like you either need to delete your pio folder or update to the newer arduino-esp32. Barring that you may want to use the 1.3.8 release instead of the master branch. I'll have to fix this for the older arduio-esp32 versions.

trevorsb, 2022-05-31:
Hey h2zero, thank you for replying! I tried deleting the `.pio` folder, but n success.

Could you give me a quick explanation of how to update to the newer arduino-esp32 or to pull back to the 1.3.8 release? I'm sorry, I know this is simple, but not my wheelhouse. Thank you!

trevorsb, 2022-05-31:
Update: I took the momentous step and googled it - updating arduino-esp32 resolved the issue!

Thank you @h2zero !

h2zero, 2022-05-31:
Lol, great!

h2zero, 2022-06-03:
Resolved in [726b405](https://github.com/h2zero/NimBLE-Arduino/commit/726b405ac45399d02250543e2e97c263467c967a)

closed 2022-06-03 by h2zero