h2zero/NimBLE-Arduino #410: esp32 ble scan not working

chrisveni, 2022-06-09:
hi 
i am new to arduino,i modified the Nimble scan code to decode the ble beacon temperature ,humidity and magnetic field detection values  .it was working for few minutes and later i got this .

Decoding stack results
0x40090bbc: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x40090ded: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x400ff3af: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 47
0x400ff3f6: std::terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 57
0x400fecf9: __cxxabiv1::__cxa_allocate_exception(std::size_t) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_alloc.cc line 268
0x400fea4c: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc line 54
0x400d220f: std::vector   >::_M_range_insert (__gnu_cxx::__normal_iterator    > >, unsigned char const*, unsigned char const*, std::forward_iterator_tag) at c:\users\janec\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\ext/new_allocator.h line 104
0x400d22b9: NimBLEAdvertisedDevice::setPayload(unsigned char const*, unsigned char, bool) at c:\users\janec\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits/stl_vector.h line 1377
0x400d344f: NimBLEScan::handleGapEvent(ble_gap_event*, void*) at C:\Users\janec\Documents\Arduino\libraries\NimBLE-Arduino\src\NimBLEScan.cpp line 114
0x400d76ae: ble_gap_rx_adv_report at C:\Users\janec\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\host\src\ble_gap.c line 989
0x400dc121: ble_hs_hci_evt_le_adv_rpt at C:\Users\janec\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\host\src\ble_hs_hci_evt.c line 501
0x400dc1cd: ble_hs_hci_evt_le_meta at C:\Users\janec\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\host\src\ble_hs_hci_evt.c line 301
0x400dc419: ble_hs_hci_evt_process at C:\Users\janec\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\host\src\ble_hs_hci_evt.c line 859
0x400dacf1: ble_hs_event_rx_hci_ev at C:\Users\janec\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\host\src\ble_hs.c line 520
0x400e1eda: nimble_port_run at C:\Users\janec\Documents\Arduino\libraries\NimBLE-Arduino\src/nimble/nimble_npl_os.h line 121
0x400d2cbf: NimBLEDevice::host_task(void*) at C:\Users\janec\Documents\Arduino\libraries\NimBLE-Arduino\src\NimBLEDevice.cpp line 728
0x4008ca95: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

here is my code //

/*
   Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleScan.cpp
   Ported to Arduino ESP32 by Evandro Copercini
*/

/** NimBLE differences highlighted in comment blocks **/

/*******original********
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>
***********************/

#include <NimBLEDevice.h>
#include <NimBLEAdvertisedDevice.h>

int scanTime = 5; //In seconds
BLEScan* pBLEScan;
//uint8_t cManufacturerData1[100];
String bledata="";
String Scaned_BLE_Address;
static BLEAddress *Server_BLE_Address;
uint8_t cManufacturerData1[100];
uint8_t cManufacturerData2[100];
uint8_t cManufacturerData3[100];
uint8_t cManufacturerData4[100];
uint8_t cManufacturerData5[100];

int temp1;
int hump1;
int mag1;
int temp2;
int hump2;
int mag2;
int temp3;
int hump3;
int mag3;
int temp4;
int hump4;
int mag4;
int temp5;
int hump5;
int mag5;
//String bledata="";

[... 236 lines cut ...]
   bledata+="*";
    bledata+=mag2;
   bledata+="*";
   bledata+=temp3*0.01;
  bledata+="*";
  bledata+=hump3;
   bledata+="*";
    bledata+=mag3;
   bledata+="*";
   bledata+=temp4*0.01;
  bledata+="*";
  bledata+=hump4;
   bledata+="*";
    bledata+=mag4;
   bledata+="*";
   bledata+=temp5*0.01;
  bledata+="*";
  bledata+=hump5;
  bledata+="*";
  bledata+=mag5;
  bledata+="#";
  Serial.println(bledata);
  
  delay(5000);
}

h2zero, 2022-06-09:
Hello @chrisveni, it looks to be a timing issue due to the lengthy callback handler you have. I believe this could be easily resolved by removing the call to `pBLEScan->clearResults();` in your loop(). This is not really necessary in this use.

chrisveni, 2022-06-12:
hello @h2zero thank you for replying ,I tried removing the pBLEScan->clearResults();  but still i am facing  the issue .

Decoding stack results
0x40090bbc: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x40090ded: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x400ff8a7: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 47
0x400ff8ee: std::terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 57
0x400ff1f1: __cxxabiv1::__cxa_allocate_exception(std::size_t) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_alloc.cc line 268
0x400fef44: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc line 54
0x400d25bb: std::vector   >::_M_range_insert (__gnu_cxx::__normal_iterator    > >, unsigned char const*, unsigned char const*, std::forward_iterator_tag) at c:\users\janec\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\ext/new_allocator.h line 104
0x400d2665: NimBLEAdvertisedDevice::setPayload(unsigned char const*, unsigned char, bool) at c:\users\janec\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits/stl_vector.h line 1377
0x400d37fb: NimBLEScan::handleGapEvent(ble_gap_event*, void*) at C:\Users\janec\Documents\Arduino\libraries\NimBLE-Arduino\src\NimBLEScan.cpp line 114
0x400d7a5a: ble_gap_rx_adv_report at C:\Users\janec\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\host\src\ble_gap.c line 989
0x400dc4cd: ble_hs_hci_evt_le_adv_rpt at C:\Users\janec\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\host\src\ble_hs_hci_evt.c line 501
0x400dc579: ble_hs_hci_evt_le_meta at C:\Users\janec\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\host\src\ble_hs_hci_evt.c line 301
0x400dc7c5: ble_hs_hci_evt_process at C:\Users\janec\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\host\src\ble_hs_hci_evt.c line 859
0x400db09d: ble_hs_event_rx_hci_ev at C:\Users\janec\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\host\src\ble_hs.c line 520
0x400e2286: nimble_port_run at C:\Users\janec\Documents\Arduino\libraries\NimBLE-Arduino\src/nimble/nimble_npl_os.h line 121
0x400d306b: NimBLEDevice::host_task(void*) at C:\Users\janec\Documents\Arduino\libraries\NimBLE-Arduino\src\NimBLEDevice.cpp line 728
0x4008ca95: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

h2zero, 2022-06-12:
Thank you for reporting back. I had a chance to test your code and now that I can see it better I have identified the issue.

You were running out of memory, as the backtrace suggested. The cause is calling `new` to create a copy of the device address in the callback. Here is the updated code with remarks.
```
/*
   Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleScan.cpp
   Ported to Arduino ESP32 by Evandro Copercini
*/

/** NimBLE differences highlighted in comment blocks **/

/*******original********
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>
***********************/

#include <NimBLEDevice.h>
#include <NimBLEAdvertisedDevice.h>

int scanTime = 5; // In seconds
BLEScan *pBLEScan;
// uint8_t cManufacturerData1[100];
String bledata = "";
String Scaned_BLE_Address;
static BLEAddress Server_BLE_Address; // Just create a BLEAddress object instead of a pointer.
uint8_t cManufacturerData1[100];
uint8_t cManufacturerData2[100];
uint8_t cManufacturerData3[100];
uint8_t cManufacturerData4[100];
uint8_t cManufacturerData5[100];

int temp1;
int hump1;
int mag1;
int temp2;
int hump2;
int mag2;
int temp3;
int hump3;
int mag3;
int temp4;
int hump4;
int mag4;
int temp5;
int hump5;
int mag5;
// String bledata="";

class MyAdvertisedDeviceCallbacks : public BLEAdvertisedDeviceCallbacks
{
    /*** Only a reference to the advertised device is passed now
      void onResult(BLEAdvertisedDevice advertisedDevice) { **/
    void onResult(BLEAdvertisedDevice *advertisedDevice)
    {
        yield();
        // Serial.printf("Advertised Device: %s \n", advertisedDevice->toString().c_str());
        Server_BLE_Address = BLEAddress(advertisedDevice->getAddress()); // Using new here to get a pointer causes a memory leak.
        Scaned_BLE_Address = Server_BLE_Address.toString().c_str(); // use dot syntax instead now
        // Serial.printf("Advertised Device: %s \n", advertisedDevice->toString().c_str());
        if (Scaned_BLE_Address == "7c:d9:f4:1a:d4:d1")
        {
            // Serial.println(Scaned_BLE_Address);
            if (advertisedDevice->haveManufacturerData() == true)
            {
                std::string strManufacturerData = advertisedDevice->getManufacturerData();

                // uint8_t cManufacturerData1[100];
                strManufacturerData.copy((char *)cManufacturerData1, strManufacturerData.length(), 0);
[... 193 lines cut ...]
    bledata += mag2;
    bledata += "*";
    bledata += temp3 * 0.01;
    bledata += "*";
    bledata += hump3;
    bledata += "*";
    bledata += mag3;
    bledata += "*";
    bledata += temp4 * 0.01;
    bledata += "*";
    bledata += hump4;
    bledata += "*";
    bledata += mag4;
    bledata += "*";
    bledata += temp5 * 0.01;
    bledata += "*";
    bledata += hump5;
    bledata += "*";
    bledata += mag5;
    bledata += "#";
    Serial.println(bledata);
    Serial.printf("Heap: %u\n", ESP.getFreeHeap()); //Added heap checking for debugging
    delay(5000);
}
```

chrisveni, 2022-06-14:
Thank you @h2zero now its working fabulously..

closed 2022-07-09 by h2zero