h2zero/NimBLE-Arduino #411: Porting code with 2901 descriptors

jackjansen, 2022-06-15:
I'm porting code where I've created 2901 descriptors for almost everything with

```
class BLE2901 : public BLEDescriptor {
public:
  BLE2901(const char *description) : BLEDescriptor(BLEUUID((uint16_t)0x2901)) { setValue((uint8_t *)description, strlen(description)); }
};

static BLE2901 levelVBat2901("Battery Level");
 ```

and then later add `pChar->addDescriptor(levelVBat2901)` .

I'm under the impression that this code will have to be restructured, but really posting this issue to ask if I've overlooked something...

h2zero, 2022-06-15:
Hi @jackjansen, sorry for the lack of documentation regarding this. Unfortunately what you're trying to port "should" work but I missed adding the default parameters to the descriptor constructor.

There are 2 options to resolve this:
1. add the extra required parameters to your construction (properties and max length) 
2. add default values to the NimBLEDescriptor constructor and submit a PR

I like #2 best :smile:

jackjansen, 2022-06-17:
Hmm. So if I understand you correctly statically allocating a NimBLEDescriptor should work? I didn't go down this path because the examples suggested I should use `pChar->createDescriptor()` in stead of creating the descriptor myself and then calling `pChar->addDescriptor()`.

h2zero, 2022-06-17:
Yes, the code has changed since the examples were made. You can create and add descriptors/characteristics outside and add them later now.

jackjansen, 2022-06-20:
I actually like the Nimble-Arduino code structure for creating the descriptors much better than my old static allocation code.

So I've converted my code, and I won't be submitting a PR.

closed 2022-06-20 by jackjansen

h2zero, 2022-06-20:
Cool, sounds good. I'll update the code later then.