h2zero/NimBLE-Arduino #412: Problem compiling

Juriaan-slash, 2022-06-17:
Hi,

When I compile a example in the Arduino IDE I get this error. What am I missing?

In member function 'virtual void DescriptorCallbacks::onWrite(NimBLEDescriptor*)':
BLE_uart:129:51: error: conversion from 'uint8_t*' {aka 'unsigned char*'} to non-scalar type 'std::__cxx11::string' {aka 'std::__cxx11::basic_string<char>'} requested
         std::string dscVal = pDescriptor->getValue();
                              ~~~~~~~~~~~~~~~~~~~~~^~
Bibliotheek NimBLE-Arduino op versie 1.3.8 in map: C:\Users\Juriaan Spek\Documents\Arduino\libraries\NimBLE-Arduino  wordt gebruikt
exit status 1
conversion from 'uint8_t*' {aka 'unsigned char*'} to non-scalar type 'std::__cxx11::string' {aka 'std::__cxx11::basic_string<char>'} requested

Thx

h2zero, 2022-06-17:
It looks like you're using the example from the master branch and compiling with the release branch. You'll need to use the examples provided with the library version you're using.

Juriaan-slash, 2022-06-17:
Yes. You are right. I'll allign and  let you know.
Thx for the quick response.

closed 2022-06-19 by Juriaan-slash