h2zero/NimBLE-Arduino #420: Allow Device Name change while BLE is running

Scoutzknifez, 2022-07-21:
As you mentioned (h2zero) in #359 :

> Do you need to change the advertised name or the device name?, they're different things.
> To change the device name you would need to call deinit/init with the new name.

I would like the capability to change the device name WITHOUT deiniting BLE.
Bluedroid currently supports this functionality in some way and it would be nice to have it here.

Scoutzknifez, 2022-07-21:
I added this:
```
void NimBLEDevice::setDeviceName(const std::string &deviceName) {
    ble_svc_gap_device_name_set(deviceName.c_str());
}
```
To NimBLEDevice.cpp and it appears to be doing what I need.
Are there any complications of doing this?

h2zero, 2022-07-21:
That's the correct solution. If you would like to PR that I would be happy to merge it.

Scoutzknifez, 2022-07-21:
Added the PR for this code in #421 
Thank you @h2zero

closed 2022-07-21 by Scoutzknifez