h2zero/NimBLE-Arduino #423: NimBLERemoteCharacteristic::unsubscribe failures starting with v1.3.8. Still in v1.4.0

devWaves, 2022-07-27:
Hi @h2zero . Let's start with a well deserved thank you for the great work you have done with this library.

I am the primary developer of the project https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32  and it would be nowhere without your NimBLE-Arduino library

Starting with v1.3.8 of NimBLE-Arduino an issue started occurring with unsubscribing to notifications. The 2 things I have noticed...

1. If I call ```pChr->unsubscribe()``` once it will cause the BLE scanning to stop calling the ```onResult(NimBLEAdvertisedDevice* advertisedDevice)``` callback, even though ```pScan->isScanning()``` returns true. I call the first unsubscribe in the notifyCB callback for the notify response

2. If I call ```pChr->unsubscribe()``` a 2nd time, it causes the ESP32 to freeze. The 2nd call is more of a backup if the first unsubscribe didn't get called if a response was not received

In my code the method I use for unsubscribing is...

```
bool unsubscribeToNotify(NimBLEClient* pClient) {
  NimBLERemoteService* pSvc = nullptr;
  NimBLERemoteCharacteristic* pChr = nullptr;

  pSvc = pClient->getService("cba20d00-224d-11e6-9fb8-0002a5d5c51b"); // custom device service
  if (pSvc) {    /** make sure it's not null */
    pChr = pSvc->getCharacteristic("cba20003-224d-11e6-9fb8-0002a5d5c51b"); // custom characteristic to notify
  }
  if (pChr) {    /** make sure it's not null */
    if (pChr->canNotify()) {
      if (!pChr->unsubscribe()) {
        return false;
      }
    }
  }
  else {
    if (printSerialOutputForDebugging) {
      Serial.println("CUSTOM notify service not found.");
    }
    return false;
  }
  if (printSerialOutputForDebugging) {
    Serial.println("unsubscribed to notify");
  }
  return true;
}
```

if I comment out the ``` if (!pChr->unsubscribe())``` part, things seem to run normally again, but I am assuming still subscribed to notifications

This all worked previously in v1.3.7 NimBLE-Arduino and is the latest version that works for me

Thanks again

h2zero, 2022-07-28:
The problem for your code is that the write to the CCCD of the characteristic has been changed to always use a write with response request. This blocks the calling task until the response is received, which is causing your issue. This was a necessary change because it's specified by the BLE standard and would cause problems with devices that do not allow a write to CCCD without response.

The solution for your issue is to not call unsubscribe from the notification callback.

devWaves, 2022-07-28:
ok thanks, that makes sense. I will try this out later today. Thanks for the quick response

h2zero, 2022-07-29:
You're welcome, let me know how you make out with it.

devWaves, 2022-07-29:
Yep it looks calling the unsubscribe from within the notify callback causes the issues. Just testing a bit more and I will release an update to my code. Cheers and thanks

closed 2022-07-29 by devWaves