h2zero/NimBLE-Arduino #424: Whitelist only calls onResult callback for max 12 devices

devWaves, 2022-07-27:
Hi @h2zero . Me again. Once again thank you for the great work you have done with this library.

I am the primary developer of the project https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32 and it would be nowhere without your NimBLE-Arduino library

I am attempting to use whitelisting and it is successful, but the onResult callback for advertised devices is only matching on max 12 devices. I am able to add 14 devices to the whitelist (When I call ```NimBLEDevice::getWhiteListCount()``` it equals 14). But only 12 of the devices advertised data is being discovered and calling the onResult callback method

I am adding to the white list like this... (anAddr = MAC address)
```NimBLEDevice::whiteListAdd(NimBLEAddress(anAddr, 1));```

and setting use whitelist mode like this...
```pScan->setFilterPolicy(BLE_HCI_SCAN_FILT_USE_WL);```

thanks

h2zero, 2022-07-28:
This may be a controller limitation, I'm not sure yet though. I will get back to you on this when I have a chance to look.

h2zero, 2022-07-31:
Just to update this, 12 whitelist devices seems to be the hard limit for the esp32.
https://github.com/espressif/esp-idf/issues/9205

devWaves, 2022-08-05:
hey @h2zero thanks for looking into this. Since the issue isnt directly related to ypur code, feel free to close the issue. Might be good to keep open for other people to be aware though

Would be nice to have more than 12 devices in whitelist, so hopefully they make an update to the base library

h2zero, 2022-08-05:
> so hopefully they make an update to the base library

Doubtful to happen any time soon, it's set in the BLE controller closed source binary to this limit, which they don't seem to keen to change parameters like this for.

closed 2024-06-04 by h2zero