h2zero/NimBLE-Arduino #425: Send data to a specific device in the characteristic

roysG, 2022-07-28:
I am trying yo send sensitive data to specific device id in the characteristic.
But as i test it, i can see the sensitive data also from others devices that connected to the same service/characteristic.

Is there any option to isolated the transferd data just for the specific device?

h2zero, 2022-07-29:
If other devices are connected and all writing to the same characteristic then there is nothing from the client side that could be done. However on the server side it may be able to save the written data and the address of the client and store it separately in from the onWrite callback.

roysG, 2022-07-29:
Ok, in case i know the adreessId of the device. do i have the ability to send just him messages?

h2zero, 2022-07-29:
Sorry, not sure if I follow what you're asking. I'm guessing the esp32 is acting as a server and you want to notify a specific client only? This could be done but the capability is not implemented in the library at this time.

JulienLecoq, 2022-08-17:
Hello guys, I'm also looking for this functionality. Notifying a single client will allow to create more easily request/response pattern. It would be of great benefit. :)

h2zero, 2022-08-17:
This should be possible without too much trouble, I'll add this as a future feature.

JulienLecoq, 2022-08-18:
Cool ! :)

h2zero, 2024-06-04:
This has been added in the master branch.

closed 2024-06-04 by h2zero