h2zero/NimBLE-Arduino #426: Package version 2.0.0+sha.a638a6d doesn't satisfy requirements ^1.4.0

alf45tar, 2022-08-01:
When using https://github.com/h2zero/NimBLE-Arduino@^1.4.0 in platformio.ini I have the following error

Library Manager: Installing git+https://github.com/h2zero/NimBLE-Arduino @ ^1.4.0
git version 2.30.1 (Apple Git-130)
Cloning into '/Users/gianluca/.platformio/.cache/tmp/pkg-installing-rvrvn3wk'...
remote: Enumerating objects: 553, done.
remote: Counting objects: 100% (553/553), done.
remote: Compressing objects: 100% (442/442), done.
remote: Total 553 (delta 73), reused 389 (delta 29), pack-reused 0
Receiving objects: 100% (553/553), 1.11 MiB | 2.44 MiB/s, done.
Resolving deltas: 100% (73/73), done.
Error: Package version 2.0.0+sha.a638a6d doesn't satisfy requirements ^1.4.0 based on PackageMetaData <type=library name=NimBLE-Arduino version=2.0.0+sha.a638a6d spec={'owner': None, 'id': None, 'name': 'NimBLE-Arduino', 'requirements': '^1.4.0', 'uri': 'git+https://github.com/h2zero/NimBLE-Arduino'}

h2zero, 2022-08-01:
For now just change your ini to remove the `^` before 1.4.0, I'll fix this after I get the PR's merged.

h2zero, 2022-08-02:
Should be resolved now, I swapped the default branch so that platformio won't look there.

closed 2022-08-03 by alf45tar