h2zero/NimBLE-Arduino #428: Add additional overloads for the method NimBLEAdvertising::setManufacturerData

VNovytskyi, 2022-08-02:
There is a problem with assigning binary data to the manufacturing field via the method:
`void NimBLEAdvertising::setManufacturerData(const std::string &data)`

For example, if execute the `setManufacturerData("\x12\x65\x67\x00\x43\x53");` the actual result is 
[image]
But the expected result must be
[image]

It occurs because the `std::string` is not intended for storing binary data and the byte `"\x00"` is perceived as the end of the string and the rest of the data is ignored.

I suggest adding new methods to solve this problem:
```c
/**
 * @brief Set the advertised manufacturer data.
 * @param [in] data The data to advertise.
 */
void NimBLEAdvertising::setManufacturerData(const std::string &data) {
    m_mfgData.assign(data.begin(), data.end());
    m_advData.mfg_data = &m_mfgData[0];
    m_advData.mfg_data_len = m_mfgData.size();
    m_advDataSet = false;
} // setManufacturerData

void NimBLEAdvertising::setManufacturerData(const std::vector<uint8_t> &data) {
    m_mfgData.assign(data.begin(), data.end());
    m_advData.mfg_data = &m_mfgData[0];
    m_advData.mfg_data_len = m_mfgData.size();
    m_advDataSet = false;
}

void NimBLEAdvertising::setManufacturerData(const uint8_t* data_p, size_t length) {
    m_mfgData = std::vector<uint8_t>(data_p, data_p + length);
    m_advData.mfg_data = &m_mfgData[0];
    m_advData.mfg_data_len = m_mfgData.size();
    m_advDataSet = false;
}
```

chegewara, 2022-08-02:
I agree it is not fortunate to use std::string for binary data, but i cant agree with your conclusion.
The problem you see is how you are using it in the code, because `std::string` does have such constructor, which allows to use 0x00 in the middle:
`std::string (const char* s, size_t n);`
and the way you are using it in code is this:
`std::string (const char* s);`

You can try:
`setManufacturerData(std::string("\x12\x65\x67\x00\x43\x53", 6));`

h2zero, 2022-08-03:
I plan to update this in the future, possibly by using `NimBLEAttValue` to leverage the templates/overloads. However, @chegewara is correct, `std::string` construction already provides the overload needed.

> I agree it is not fortunate to use std::string for binary data

I think it was brilliant really :smile:. I never considered doing that before until I saw it in esp32_snippets, it's just an extended `std::vector<uint8_t>` container after all, which is quite handy. Maybe just a bit too much overhead but quite functional.

VNovytskyi, 2022-08-03:
It will be great @h2zero! I will be waiting for this update! 😁
> I plan to update this in the future, possibly by using NimBLEAttValue to leverage the templates/overloads.

And I also agree with @chegewara that we can use now:
> setManufacturerData(std::string("\x12\x65\x67\x00\x43\x53", 6));

closed 2022-08-03 by VNovytskyi