h2zero/NimBLE-Arduino #429: SetValue not matching Function to call when update from 1.3.8 to 1.4.0

SosaL08, 2022-08-03:
I used to set the value of my characteristics directly using a char array variable at the SetValue() function.
Now, once I updated to 1.4.0 version of the library, the code can't compile cause the variable I use it's not in a valid format.
How do I have to reference or address my char array variable to be compatible with this version?

h2zero, 2022-08-03:
Can you share the code you're using?

SosaL08, 2022-08-03:
```
NimBLECharacteristic *pReadCharacteristic;
pReadCharacteristic = pBleService->createCharacteristic("A002", NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::NOTIFY);
pReadCharacteristic->setCallbacks(&chrCallbacks);
```
```
         char buffer_response[jsonWiFi.length() + 1];
          jsonWiFi.toCharArray(buffer_response, jsonWiFi.length() + 1);
          pReadCharacteristic->setValue(buffer_response);
          pReadCharacteristic->notify();
```
I can't share the entire code but this is the part that gives me trouble. When I run it with versio 1.3.8 it works perfectly, yet, updating to 1.4.0 gives me an error with the SetValue function. It says the value I'm providing its not correctly expressed, as I can't set it directly from my char array.

h2zero, 2022-08-03:
Thank you for the report and sharing the code to reproduce this. The templates do not seem to be handling this use case properly at the moment, I will look into this further for the next release. For now you can workaround the problem by casting your array to `uint8_t*`.

`pReadCharacteristic->setValue((uint8_t*)buffer_response);`

SosaL08, 2022-08-04:
Thanks! I'll try that out to see if it works correctly with my code

HankLloydRight, 2022-12-30:
I had this same problem upgrading from 1.3.7 to 1.4.1 and simply casting the char * to (uint8_t*) worked in the setValue call. Thanks.

h2zero, 2026-04-10:
Closing this as resolved

closed 2026-04-10 by h2zero