h2zero/NimBLE-Arduino #434: undefined reference to `xPortEnterCriticalTimeout'

AndrewMagpie, 2022-08-10:
I'm new to using nimble, but was using default esp32 BLE library authored by kolban.

When I try build my sketch I get the follow compiler errors:

h2zero, 2022-08-10:
Interesting, I have not seen this before. Can you share the build environment and settings used?

AndrewMagpie, 2022-08-11:
Windows 10 environment
Arduino 1.8.19
ESP32 - Espressif v2.0.3  (v2.0.4 doesnt seem to work for me)
Using these libraries:
  https://github.com/lbernstone/arduino-esp32-solo
  https://github.com/Bodmer/TFT_eSPI)
  https://github.com/lucadentella/TOTP-Arduino)
  https://github.com/arduino-libraries/NTPClient)

h2zero, 2024-06-13:
Closing this as an environment error, not an issue with this library.

closed 2024-06-13 by h2zero