h2zero/NimBLE-Arduino #435: Advertised Device without MAC address

alberk8, 2022-08-16:
I have used the NimbeBLE_Scan_Continuous example and getting the following output.  The first device actually has MAC address when scanned using NrfConnect on android.   May I know what might be the problem here?  Thank you.

```
Advertised Device: Name: TAI T8897 
Advertised Device: Name: , Address: fe:93:42:0f:16:f9, manufacturer data: 4c0012020000 
Advertised Device: Name: , Address: 7b:74:07:26:0a:92, manufacturer data: 4c000100000000000000000000000000000100, txPower: 7 
Advertised Device: Name: , Address: ad:6e:18:1a:89:1c
Service Data:
UUID: 0xfef1, Data: =⸮ ⸮޸:P92⸮⸮⸮⸮GE;c
```

h2zero, 2022-08-16:
Weird... what does nRFConnect show? Just a guess but the device's name may have a terminating`'\0'` which could cause this.
You could test by changing the code and printing the address separately:
`Serial.printf("Address: %s\n", advertisedDevice->getAddress().toString().c_str());`

alberk8, 2022-08-16:
> Serial.printf("Address: %s\n", advertisedDevice->getAddress().toString().c_str());

Yes, I can confirm that with printing the MAC address it does show.  So it could be the name is null terminated.

h2zero, 2022-08-16:
Yes the device is broadcasting it's name with a terminating NULL, which produces the result you are seeing. Pretty amusing really, but annoying for sure (and a waste of a byte in the advertising space). For this device, this is your only workaround option, you'll have to manually fetch/print the data you want from it.

closed 2022-08-16 by alberk8