h2zero/NimBLE-Arduino #437: Crash when WiFi modem sleep is disabled before BLE initialization

thorrak, 2022-08-19:
Something I noticed recently while trying to debug mDNS issues in one of my projects is that if WiFi modem sleep is disabled, either via the ESP-IDF function `esp_wifi_set_ps(WIFI_PS_NONE);` or one of the variants of the the Arduino function `WiFi.setSleep(false);` then initializing NimBLE crashes with an error similar to the following:

```
abort() was called at PC 0x4017f830 on core 1

Backtrace:0x40083fb1:0x3ffd35e00x40095e35:0x3ffd3600 0x4009bae1:0x3ffd3620 0x4017f830:0x3ffd36a0 0x4017f65b:0x3ffd36c0 0x4013e261:0x3ffd36e0 0x400f48f2:0x3ffd3700 0x400decb5:0x3ffd3750 0x400d95c9:0x3ffd3790 0x401101f6:0x3ffd37b0 

  #0  0x40083fb1:0x3ffd35e0 in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:402
  #1  0x40095e35:0x3ffd3600 in esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c:128
  #2  0x4009bae1:0x3ffd3620 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/abort.c:46
  #3  0x4017f830:0x3ffd36a0 in coex_core_enable at ??:?
  #4  0x4017f65b:0x3ffd36c0 in coex_enable at ??:?
  #5  0x4013e261:0x3ffd36e0 in esp_bt_controller_enable at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/controller/esp32/bt.c:1848
  #6  0x400f48f2:0x3ffd3700 in NimBLEDevice::init(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) at .pio/libdeps/lcd_ssd1306/NimBLE-Arduino/src/NimBLEDevice.cpp:770
```

Digging in, the specific line in esp-idf which is causing the crash is [this one](https://github.com/espressif/esp-idf/blob/v4.4.2/components/bt/controller/esp32/bt.c#L1848) enabling software coexistence.

Interestingly, there is actually an error message logged if the WiFi modem sleep is disabled _after_ NimBLE initialization, following BLE activity and immediately preceding a crash:

`E (10532) wifi:Error! Should enable WiFi modem sleep when both WiFi and Bluetooth are enabled!!!!!!`

I'm not sure what the best solution to this would be -- there [is a function](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv415esp_wifi_get_psP14wifi_ps_type_t) `esp_wifi_get_ps` to query the "sleep state" that returns [a value](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv414wifi_ps_type_t) that can be tested against. The value that will guarantee a crash is `WIFI_PS_NONE` -- the default value (and the one that you want for software coexistence) is `WIFI_PS_MIN_MODEM`. That said, I don't know what the effect would be of querying this if coexistence isn't needed/available in a sketch -- an alternative would be just adding a comment to the code referencing the power saving functions as a possible cause of a crash at the [specific line](https://github.com/h2zero/NimBLE-Arduino/blob/release/1.4/src/NimBLEDevice.cpp#L880) that the trace references.

h2zero, 2022-08-19:
Thanks! 

This is probably something that should be in the docs somewhere. 

I don't really want to add any code to check this setting to the library, it's better for the user to be aware of the requirements instead.

thorrak, 2022-08-19:
Yeah — it feels to me like something that should be addressed in esp-idf, somewhere in coex_enable(), either by raising an error, changing the sleep mode, or both. I went ahead and raised an issue over there for this: https://github.com/espressif/esp-idf/issues/9595

closed 2024-06-04 by h2zero