h2zero/NimBLE-Arduino #440: Set Device Name

Ozark-iot, 2022-08-25:
I am using Nordic Semi nRF52840 to advertise in Coded PHY then other devices are scanning for the Device name set within the advertising packet as shown below.

What would help me a huge amount is if you would be kind enough to show me how to configure a device name within a Coded PHY advertising packet created by NimBLE-Arduino

Sorry to be a nuisance for asking :)

#define DEVICE_NAME CONFIG_BT_DEVICE_NAME
#define DEVICE_NAME_LEN (sizeof(DEVICE_NAME) - 1)
static struct bt_le_ext_ad * ext_adv;

static struct bt_le_adv_param *adv_param_codedphy =
       BT_LE_ADV_PARAM(BT_LE_ADV_OPT_USE_IDENTITY | BT_LE_ADV_OPT_USE_TX_POWER | BT_LE_ADV_OPT_CODED | 
       BT_LE_ADV_OPT_EXT_ADV,
       BT_GAP_ADV_FAST_INT_MIN_1, BT_GAP_ADV_FAST_INT_MAX_1,NULL);        

#define Ozark_Advert_Data_Length 70
#define MANUFACTURER_ID          0x0059

static uint8_t Advert_Ozark[Ozark_Advert_Data_Length];

static struct bt_data ad[] = {
	BT_DATA(**BT_DATA_NAME_COMPLETE, DEVICE_NAME, DEVICE_NAME_LEN**),
        BT_DATA(BT_DATA_MANUFACTURER_DATA, Advert_Ozark, Ozark_Advert_Data_Length),
};

h2zero, 2022-08-25:
I'm not sure about the code you've provided above, but in this library the call structure would be:

```
NimBLEExtAdvertisement extAdv(primaryPhy, secondaryPhy);
extAdv.setName(DEVICE_NAME);

NimBLEExtAdvertising* pAdvertising = NimBLEDevice::getAdvertising();
pAdvertising->setInstanceData(0, extAdv);
pAdvertising->start(0, advTime);
```

Ozark-iot, 2022-08-25:
Thank you so much, extremely helpful.
My understanding is that the following call forces a reboot, is there a more delicate way of doing the following:

Stop advertising
Delete advertisment
Create advertisment
Start advertising

esp_sleep_enable_timer_wakeup(sleepSeconds * 1000000);

h2zero, 2022-08-25:
There is no way to avoid the reboot if you need to use deep sleep. If you want to use light sleep I think you can just deinit the BLE/WiFi stack and enter light sleep then re-init etc. on wake-up. I can't say for sure though if that will work as I have not tried it.

Ozark-iot, 2022-08-25:
Thank you once again for your kind help.

Ozark-iot, 2022-08-25:
One more thing.

Are we able to dynamically change the advertising packet?
Ignoring any sleep functionality.

h2zero, 2022-08-25:
Yes, as long as you stop advertising first. After that you can change whatever you like in the same way you originally set it.

Ozark-iot, 2022-08-25:
As mentioned on Gitter, I have set the name correctly I believe, however I'm unable to see any transmissions, any help would be very much appreciated.

closed 2024-06-04 by h2zero