h2zero/NimBLE-Arduino #442: 1.4.0 maybe causes Guru Meditation Error: Cache disabled but cached memory region accessed together with NVS

immakermatty, 2022-08-28:
Hello, I have a big PlatformIO project for ESP32 that uses NimBLE-Arduino. I am mixing WiFi, ESP-NOW, TWAI, Neopixels and BLE.  After the latest 1.4.0 I suddenly started to get once in a while _Guru Meditation Error: Cache disabled but cached memory region accessed_ crashes when I try to write to NVS. After some poking around I isolated the source of the problem in the recent update of the NimBLE-Arduino library. The problem was fixed by downgrading to 1.3.8

I am on the [5.0.0 espresiff32 platform](https://github.com/platformio/platform-espressif32/releases/tag/v5.0.0) (ESP-IDF 4.4.1 I think)

`esp_err_t err;
nvs_handle my_handle;
if ((err = nvs_open("project", NVS_READWRITE, &my_handle)) == ESP_OK) {

    if ((err = nvs_set_u16(my_handle, "ota_ver", version)) != ESP_OK) {  // <-- crashes on this line
        log_e("Error setting product code (%s)", esp_err_to_name(err));
    }

    if ((err = nvs_commit(my_handle)) != ESP_OK) {
        log_e("Error commiting nvs (%s)", esp_err_to_name(err));
    }
    nvs_close(my_handle);
    
} else {
    log_e("Error (%s) opening NVS handle!", esp_err_to_name(err));
}`

immakermatty, 2022-08-28:
After some programming it happened to me again with version 1.3.8. This means it is not strictly 1.4.0 related. But it definitely helped to downgrade! Maybe my IRAM is full? So not all IRAM_ATTR functions are getting to IRAM?

h2zero, 2022-08-28:
I can't think of anything in the C++ code that would affect this. It could be due to upstream changes that were applied to the NimBLE core files. I would need to have some code to reproduce this in order to diagnose it.

h2zero, 2022-08-28:
By any chance are you using SPI (maybe in neopixels)?

immakermatty, 2022-08-30:
I'm using RMT controller for the neopixels (modified version of the FastLED implementation). But good point, I'll look for any code that uses SPI.

This is the crash report, when it happends btw

> Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed).
> 
> Core  1 register dump:
> PC      : 0x40008150  PS      : 0x00060035  A0      : 0x80081568  A1      : 0x3ffbf16c
> A2      : 0x3f40364f  A3      : 0x00000000  A4      : 0x00000004  A5      : 0x00000000
> A6      : 0x00000000  A7      : 0x3ffd6e8c  A8      : 0x0000a02c  A9      : 0x00000004
> A10     : 0x000000ba  A11     : 0x00000000  A12     : 0x3ffd6df0  A13     : 0x3ffd6dd0
> A14     : 0x3ffc0454  A15     : 0x00000000  SAR     : 0x00000018  EXCCAUSE: 0x00000007
> EXCVADDR: 0x00000000  LBEG    : 0x40091564  LEND    : 0x4009157a  LCOUNT  : 0xffffffff
> 
> 
> Backtrace:0x4000814d:0x3ffbf16c |<-CORRUPTED

Exception decoder decodes nothing from the backtrase :(

immakermatty, 2022-08-30:
I found it! It was all caused by LittleFS in the end. I took the time and checked all the changes in libraries and code I've done over the past week and I found out that I changed an order to a function that is writing a file to LittleFS. Strangest part is that 

Initializing LittleFS, setting up my aplication, then write a file and then writing to nvs (all one after the other) is fine

but 

Initializing LittleFS, writing a file, setting up the application and then writing to nvs, the write to the nvs crashes.

And as the issue did not surface in the flashing of the ESP over USB, I discovered it after some days as I was doing an OTA update, shortly after I've updated NimBLE library. 

Sorry for the irellevant issue.

closed 2022-08-30 by immakermatty

h2zero, 2022-08-30:
Thanks for the update, glad you found the issue!