h2zero/NimBLE-Arduino #447: ESP32-GRBL dynamic values

leo1988joseph, 2022-09-13:
Hi, I would like to know when I flash the dbring library to ESP32 and configure it with a board (Diamond Chimp), it works when I connect with the grbl controller in Android connected via BLE.
I want to know when I send a Gcode via BLE to the ESP32, which file or exact line receives the value so that it moves the motor? Which variable can accept the value (x,y coordinates) to move the motor and in which file?

Thanks

h2zero, 2022-09-13:
Assuming the esp32 is operating as a BLE server all you need is a writable characteristic. When the client writes the data to this your code will get a callback where you can then handle the new data.

leo1988joseph, 2022-09-13:
[image]

This is the setup I am using. I basically want the coordinates of x and y axis to be provided from my phone via BLE. But I am not getting the file to were the coordinate has to be provided.

closed 2024-06-04 by h2zero