h2zero/NimBLE-Arduino #448: Error While Compiling for adafruit_feather_nrf52832

muhdRiz07, 2022-09-16:
I modified the platform.ini file as per the instruction given in readme.md file but giving error 'board_bootloader to be specified'.(refer screenshot 1 & 2)
when i add board_bootloader = adafruit it's giving library error(screen shot 3 & 4)
[image]
[image]
[image]
[image]
**board:adafruit_feather_nrf52832
   ide: platformio**

h2zero, 2022-09-17:
Hi @muhdRiz07, the problem here is cause by the line `board_build.mcu = nrf52832`. You do not need to specify this as it's defined in the board by the platform. However, if you want to define it the correct way is `board_build.mcu = nrf52832_xxaa`, the reason is this value is used to set a compile flag.

Note that adafruit boards require specifying the bootloader to use, if you're not using one you should set `board_bootloader = none`

muhdRiz07, 2022-09-22:
Hi @h2zero, now it's working,**Thanks** for the support.

closed 2022-09-22 by muhdRiz07