h2zero/NimBLE-Arduino #451: Sending longer payload -> How to set (request) MTU change from server side

marcelrv, 2022-09-20:
I'm trying to send a larger characteristic than the default 20 bytes.
I'm trying to understand how I can request a MTU change from the server side.

See below NimBLE_Server  example modified to (how I understood) request the larger MTU

added near the  NimBLEDevice::init    `NimBLEDevice::setMTU(48);`
added in the onConnect `pServer->setDataLen(desc->conn_handle, 48);`
and changed the payload to be longer

Neither seemed to work.
If I change the MTU from the client side it works very well.

note: I saw in the `NimBLEServer::setDataLen(uint16_t conn_handle, uint16_t tx_octets)` that I needed at least ESP_IDF_VERSION 432.
I indeed updated it and now I have ESP_IDF_VERSION 441

server: ESP32
client: Android (nRF Connect App)

Any suggestion?

modfied example:
```
#include <Arduino.h>

/** NimBLE_Server Demo:
 *
 *  Demonstrates many of the available features of the NimBLE server library.
 *
 *  Created: on March 22 2020
 *      Author: H2zero
 *
*/

#include <NimBLEDevice.h>

static NimBLEServer* pServer;

/**  None of these are required as they will be handled by the library with defaults. **
 **                       Remove as you see fit for your needs                        */
class ServerCallbacks: public NimBLEServerCallbacks {
    void onConnect(NimBLEServer* pServer) {
        Serial.println("Client connected");
        Serial.println("Multi-connect support: start advertising");
        NimBLEDevice::startAdvertising();
    };
    /** Alternative onConnect() method to extract details of the connection.
     *  See: src/ble_gap.h for the details of the ble_gap_conn_desc struct.
     */
    void onConnect(NimBLEServer* pServer, ble_gap_conn_desc* desc) {
        Serial.print("Client address: ");
        Serial.println(NimBLEAddress(desc->peer_ota_addr).toString().c_str());
        /** We can use the connection handle here to ask for different connection parameters.
         *  Args: connection handle, min connection interval, max connection interval
         *  latency, supervision timeout.
         *  Units; Min/Max Intervals: 1.25 millisecond increments.
         *  Latency: number of intervals allowed to skip.
         *  Timeout: 10 millisecond increments, try for 5x interval time for best results.
         */
        pServer->updateConnParams(desc->conn_handle, 24, 48, 0, 60);
       // ADDED this line to change data len after the connection is made
        pServer->setDataLen(desc->conn_handle, 48);

    };
    void onDisconnect(NimBLEServer* pServer) {
        Serial.println("Client disconnected - start advertising");
        NimBLEDevice::startAdvertising();
    };
    void onMTUChange(uint16_t MTU, ble_gap_conn_desc* desc) {
        Serial.printf("MTU updated: %u for connection ID: %u\n", MTU, desc->conn_handle);
    };
[... 189 lines cut ...]
    pAdvertising->addServiceUUID(pBaadService->getUUID());
    /** If your device is battery powered you may consider setting scan response
     *  to false as it will extend battery life at the expense of less data sent.
     */
    pAdvertising->setScanResponse(true);
    pAdvertising->start();

    Serial.println("Advertising Started");
}

void loop() {
  /** Do your thing here, this just spams notifications to all connected clients */
    if(pServer->getConnectedCount()) {
        NimBLEService* pSvc = pServer->getServiceByUUID("BAAD");
        if(pSvc) {
            NimBLECharacteristic* pChr = pSvc->getCharacteristic("F00D");
            if(pChr) {
                pChr->notify(true);
            }
        }
    }

  delay(2000);
}
```

h2zero, 2022-09-20:
Normally it's expected that the client requests the MTU exchange. There is no server method for this implemented at this time but it should be possible to use the lower level calls to do this.

marcelrv, 2022-09-24:
Thanks.. will see if I can get it to work. thx

spilz87, 2023-01-30:
Hello
Did you find a way to increase it ?
Thanks in advance

h2zero, 2024-07-02:
For anyone else that finds this, the BLE specification states the MTU exchange is to be performed by the client:
[image]

The default MTU for all BLE devices is 23, 20 usable bytes. The client and server both need to have their own MTU size set higher in order to increase this. After the MTU exchange the MTU will be the smaller of the 2 values that the client and server possess.

The MTU is the max attribute (think characteristic value) size and is used for things like notifications. 
The data length sets how much of that data can be sent in a single transmission during a connection interval, setting this to a higher value than the MTU will result in greater throughput, if it's smaller, then the data will be chunked and sent slower.