h2zero/NimBLE-Arduino #452: When updating directly from 1.33 to 1.40, error. Solved by 1.33->1.38, then 1.38->1.40

CharlesRDavis, 2022-09-20:
Hi,

Since I have a workaround, this is mainly an FYI.

I update directly from 1.33 to .1.40 and get this

/Users/xxxxx/Documents/RaspberryPi/Arduino/libraries/NimBLE-Arduino/src/nimble/nimble_port.h:27:27: error: 'CONFIG_BT_NIMBLE_TASK_STACK_SIZE' undeclared (first use in this function)
 #define NIMBLE_STACK_SIZE CONFIG_BT_NIMBLE_TASK_STACK_SIZE
                           ^
/Users/xxxxx/Documents/RaspberryPi/Arduino/libraries/NimBLE-Arduino/src/porting/npl/freertos/src/nimble_port_freertos.c:49:50: note: in expansion of macro 'NIMBLE_STACK_SIZE'
     xTaskCreatePinnedToCore(host_task_fn, "ble", NIMBLE_STACK_SIZE,
                                                  ^
/Users/xxxxx/Documents/RaspberryPi/Arduino/libraries/NimBLE-Arduino/src/nimble/nimble_port.h:27:27: note: each undeclared identifier is reported only once for each function it appears in
 #define NIMBLE_STACK_SIZE CONFIG_BT_NIMBLE_TASK_STACK_SIZE
                           ^
/Users/xxxxx/Documents/RaspberryPi/Arduino/libraries/NimBLE-Arduino/src/porting/npl/freertos/src/nimble_port_freertos.c:49:50: note: in expansion of macro 'NIMBLE_STACK_SIZE'
     xTaskCreatePinnedToCore(host_task_fn, "ble", NIMBLE_STACK_SIZE,

If I update to 1.38, it still compiles. If I then update to 1.40, it compiles OK.

thorrak, 2022-09-20:
That's exceptionally strange. Are you using Arduino library manager or Platformio?

CharlesRDavis, 2022-09-20:
Arduino. Three different macs, all Mojave.

h2zero, 2022-09-21:
Interesting, not sure how this has occurred. Obviously it's due to the recent macro name change but it should not have affected an Arduino environment like this.

CharlesRDavis, 2022-09-21:
I have noticed that when I first compile something with Arduino, it takes a long time. Then subsequent compiles often are quite quick. Could this be some sort of caching issue?  Precompiled stuff? I don't know enough about the inner working of Arduino libraries to intelligently comment. One thing I did not do was restart Arduino and try. 

Very specifically, I have three Macs loaded with Arduino and keep them synced. All three were on 1.33. I updated two of them directly to the latest version, 1.40. The sketch was unchanged. The compiled failed. I went to the third computer, still on 1.33, and it compiled. 

I then went back to the first computer with 1.40, and reloaded 1.33. It compiled. I then loaded newer and new versions thinking it would eventually fail. But they all worked up to and including 1.40. 

I decided to see if I could duplicate this so that I could report it.  I went to the second computer that had 1.40. It also would not compile. I then loaded 1.33. It compiled. then 1.38. It compiled. Then 1.40. It compiled.

The third computer was on 1.33 and compiling OK. I loaded 1.40 and it failed. I then loaded 1.38 and it compiled, and, again, upon loading 1.40, it compiled.

You can look at my code if you want. Or maybe since this seem to occur only to people that don't keep up with the latest version (i.e. me), and since there is an easy workaround, it can be left alone. Up to you. 

Charles Davis

h2zero, 2022-09-21:
> I have noticed that when I first compile something with Arduino, it takes a long time. Then subsequent compiles often are quite quick. Could this be some sort of caching issue?

Yes, each time you restart Arduino it creates a new build folder which recompiles everything.

> One thing I did not do was restart Arduino and try.

I suspect this may have resolved the issue due to the caching. 

Thanks for the report though, it's certainly useful information for others. I'm not sure if anything could be done to "fix" this other than the information provided here.

CharlesRDavis, 2022-09-21:
I should also mention that I am very happy and very impressed with NimBLE. It replace the "stock" BLE on my project and a vast improvement.

closed 2022-09-21 by CharlesRDavis