h2zero/NimBLE-Arduino #456: std::string and char* are sent differently (c-strings include the null terminator)

j4m3s, 2022-09-25:
I'm not sure if this is a bug or not, but it wasn't what I expected.  

When I send a char*, the value ends with the null terminator (\u0000).  When I send a std::string, it does not.  I would expect both to send the same characters.

For example:
```
NimBLECharacteristic::setValue("hello");
NimBLECharacteristic::notify();
``` 
sends `68-65-6C-6C-6F-00`

whereas
```
NimBLECharacteristic::setValue(std::string("hello"));
NimBLECharacteristic::notify();
``` 
sends `68-65-6C-6C-6F` (note the lack of `-00`!).

I noticed because I have been using my own "terminator", `\n` in order for the client to know if the string is very and hasn't been fully received.  This worked fine until I used a c-string in one notification, and ended up with a `\u0000` AFTER my `\n` terminator.  Which was rather tricky for me to track down because, of course, the null terminator doesn't print out in the logs on the client so didn't show up :)

I'm happy for you to close the issue as invalid if this is working as intended, but thought I'd record it here for others.

h2zero, 2022-09-25:
Thanks! These 2 methods should have the same result, so certainly this is not intended. I think an adjustment to remove the trailing NULL would be best, let me know your thoughts.

j4m3s, 2022-09-25:
I agree, I'd remove the null. The end of a string at the client end will be handled by the programming language at the client end without us needing to send it :)

This _could_ be a breaking change I guess, if anyone is only ever sending char* values and is relying on that null character being there. Might be worth mentioning in the change log?

h2zero, 2022-09-25:
Yes it certainly should be noted as a possible breaking change. Thanks!

closed 2026-03-30 by h2zero