h2zero/NimBLE-Arduino #457: Cannot getCharacteristic() using NimBLEUUID

kinsi55, 2022-09-26:
For some reason, when I try to pass a `NimBLEUUID` to `getCharacteristic` it cannot get the characteristic (null returned) - Passing a string instead, works.

Passing a UUID to `getService` however, works.

String example: `pSvc->getCharacteristic("FFE1")`
UUID Example:
```
static const NimBLEUUID JK_BMS_CHARACTERISTIC_UUID(0xFFE1u);
pSvc->getCharacteristic(JK_BMS_CHARACTERISTIC_UUID);
```

ILFforever, 2023-01-13:
Hello I'm having a similar issue on an ESP32 S3 connecting to a uECG device. The board gets stuck on
 `pChr = pSvc->getCharacteristic(NimBLEUUID("FC7A850D-C1A5-F61F-0DA7-9995621FBD00"));` I'm trying to test out the method you used but since our service uuid is in a diffrent format it doesn't seems to work. Could you give me some help with this please?

h2zero, 2023-01-20:
Not sure what's going on with that but I'll look into it.