h2zero/NimBLE-Arduino #458: Nimble init gets stuck if called outside of main loop

Baby1991, 2022-10-04:
Ive been working for quite some time with this library and it worked really well. But when i tried adding multitasking to my project (ESP32) with RTOS and turning BLE on and off from a non main task it started making problems. So, when i erase the flash and upload my code everything works well on first few ble inits (through resets), but about on lets say third initialization it just gets stuck on `I NimBLEDevice: BLE Host Task Started` doing nothing and drawing a lot of power. I tried adding more task memory but i think it doesnt help. It literally hangs on `NimBLEDevice::init(ID.c_str());` where ID is just a string generated from a literal and mac address (which didnt make problems before).

h2zero, 2022-10-04:
I can't think of any reason for this behavior. Can you share a minimal code example to reproduce? Also please state the chip and NimBLE/arduino versions.

Baby1991, 2022-10-05:
Im using ESP32C3 with Arduino 2.0.3 framework although through PlatformIO, and by using it as idf component. Everything compiles okay and even deiniting turns the power down (tho it does disconnect me from wifi but thats a problem for later). Ill supply the code for it down bellow but today the problem became even worse, it doesnt even give me the BLE Host task started message, just hangs the processor doing nothing. Currently the code calls the init functon from main 
``if (turnOnBLE) {
    log_i("Turning on BLE");
    ble_init();
    BLEon = true;
    turnOnBLE = false;
  }``
  And it only says `loop(): Turning on BLE` and hangs. Clearing flash usually helps it for some time, after which i had to move it to main task but its even making problems there now. Tho in main resetting helps i think, because it starts it okay after restart. Is it maybe a problem that it isnt the highest priority task, being that C3 is single cored?

Baby1991, 2022-10-06:
@h2zero forgot to tag you in the last comment, sorry... I forgot to say that im using the latest version of the library aswell. I tried changing stack sizes and the problem persists, could it be possible that WiFi Handler function that im using blocks BLE initialization maybe?

h2zero, 2022-10-06:
Hmm, that's certainly a possibility. You could try preventing WiFi use until initialization is complete.

Baby1991, 2022-10-09:
@h2zero So after more testing ive come to 2 conclusions. It has nothing to do with wifi, because when i only start Nimble same thing happens, so its something stored in flash. Secondly, if i modify default arduino menuconfig from BlueDroid to Nimble it works but it has a lot of redefine warnings at compilation. On a passing note i think changing that messed up something else in my code because im getting a breakpoint exception out of nowhere (Do you maybe know what could be causing that?). Atleast BLE now works. Thanks again

h2zero, 2022-10-10:
Oh, wow I didn't think you were using this repo directly with esp-idf. I have not tried to do that for years, esp-nimble-cpp was designed for this since the NimBLE stack already exists there.

Baby1991, 2022-10-10:
@h2zero Oh, i tried using the cpp version but just went back to the arduino one as i transfered this project from pure arduino and it still worked. But ill look more into it now. Before i close this issue, do you maybe know where Breakpoint exceptions come from?

Baby1991, 2022-10-10:
@h2zero This may not have a place here but i wanna check before i make a new issue in esp-nimble-cpp. I tried compiling my code replacing the arduino library with the cpp one and im getting an error that `esp32-hal-bt.h` doesnt exist, although everything worked just 5 minutes before. Is that maybe a problem with C3?

Baby1991, 2022-10-10:
@h2zero bluetooth is enabled and nimble is selected. I think the problem is, as im using arduino as component in PlatformIO it compiles esp-nimble-cpp first and as such it cant reference arduino HALs

h2zero, 2022-10-10:
Hmm that sounds like a platformio issue then. esp-nimble-cpp + Arduino using IDF works just fine.

Baby1991, 2022-10-10:
@h2zero Ah I see, Ill try to make something work. Ill close this issue now. Thank you for your help!

closed 2022-10-10 by Baby1991