h2zero/NimBLE-Arduino #464: When device is paired on reconnect of the device not working

JuanJoseMoralesCalvo, 2022-10-16:
Hi im trying to create an HID Gamepad so when i first pair the device all works great but when i (for example) reset the module it doesnt work. Any suggestion on the connect CallBack?

JuanJoseMoralesCalvo, 2022-10-16:
I provide code here:
#include "NimBLEServer.h"
#include "NimBLEHIDDevice.h"
#include "NimBLEDevice.h"
bool device_connected = false;
static NimBLEHIDDevice* HID_DEVICE;
NimBLECharacteristic* REPORT;

const uint8_t reportMap[] = {
        0x05, 0x01,  /* USAGE_PAGE (Generic Desktop)       */
        0x09, 0x05,  /* USAGE (Game Pad)                   */
        0xa1, 0x01,  /* COLLECTION (Application)           */
        0xa1, 0x03,  /*   COLLECTION (Report)              */
        0x85, 0x01,  /*     REPORT_ID (1)                  */
        0x05, 0x09,  /*     USAGE_PAGE (Button)            */
        0x19, 0x01,  /*     USAGE_MINIMUM (Button 1)       */
        0x29, 0x0e,  /*     USAGE_MAXIMUM (Button 14)      */
        0x15, 0x00,  /*     LOGICAL_MINIMUM (0)            */
        0x25, 0x01,  /*     LOGICAL_MAXIMUM (1)            */
        0x95, 0x0e,  /*     REPORT_COUNT (14)              */
        0x75, 0x01,  /*     REPORT_SIZE (1)                */
        0x81, 0x02,  /*     INPUT (Data,Var,Abs)           */
        0x95, 0x01,  /*     REPORT_COUNT (1)               */
        0x75, 0x02,  /*     REPORT_SIZE (2)                */
        0x81, 0x03,  /*     INPUT (Cnst)                   */
        0xc0,        /*   END_COLLECTION                   */
        0xc0         /* END_COLLECTION                     */
    };
class ServerCallbacks: public NimBLEServerCallbacks {
    void onConnect(NimBLEServer* pServer, ble_gap_conn_desc* desc) {
        device_connected = true;
        Serial.println("Emparejado");
        if(digitalRead(17) == 1){
          Serial.println("Dentro del IF");
        NimBLEDescriptor *hid_desc = REPORT->getDescriptorByUUID(NimBLEUUID((uint16_t)0x2902));
        uint8_t val[] = {0x01, 0x00};
        hid_desc->setValue(val, 2);
        }
        pServer->updateConnParams(desc->conn_handle, 6, 8, 0, 60);
    };
    void onDisconnect(NimBLEServer* pServer) {
        device_connected = false;
        //NimBLEDevice::startAdvertising();
    };
};
void setup() {
  Serial.begin(115200);
  pinMode(17, INPUT_PULLDOWN);
  NimBLEDevice::init("Porsche922");
  NimBLEServer *server = NimBLEDevice::createServer();
  server->setCallbacks(new ServerCallbacks());

  HID_DEVICE = new NimBLEHIDDevice(server);
  HID_DEVICE->reportMap((uint8_t*)reportMap, sizeof(reportMap));
  REPORT = HID_DEVICE->inputReport(1); 

  HID_DEVICE->hidInfo(0x00,0x01);
  HID_DEVICE->manufacturer()->setValue("Porsche TAG");
  HID_DEVICE->pnp(0x03, 0x0754, 0xe123, 0x0155);

  uint8_t a[] = {0x00,0x00,0x00,0x00};
  REPORT->setValue(a, sizeof(a));    
  HID_DEVICE->startServices();

  NimBLEAdvertising *ADVERTASING_MANDATORY = server->getAdvertising();
  ADVERTASING_MANDATORY->setAppearance(HID_GAMEPAD);    // <-- optional
  ADVERTASING_MANDATORY->addServiceUUID(HID_DEVICE->hidService()->getUUID());
  ADVERTASING_MANDATORY->start();

  pinMode(18, INPUT_PULLDOWN);
  pinMode(13, INPUT_PULLDOWN);
  pinMode(27, INPUT_PULLDOWN);
  pinMode(23, INPUT_PULLDOWN);
  pinMode(14, INPUT_PULLDOWN);
  pinMode(15, INPUT_PULLDOWN);
  pinMode(16, INPUT_PULLDOWN);
  pinMode(21, INPUT_PULLDOWN);
  pinMode(33, INPUT_PULLDOWN);
  pinMode(2, INPUT_PULLDOWN);
  pinMode(4, INPUT_PULLDOWN);
  pinMode(12, INPUT_PULLDOWN);
  pinMode(22, INPUT_PULLDOWN);
  pinMode(19, INPUT_PULLDOWN);

    delay(3000);
 }

void loop() {

  if(device_connected)
  {
      uint16_t resulte = ((((GPIO.in >> 18) & 0x1)<< 0)|(((GPIO.in >> 13) & 0x1)<< 1)|(((GPIO.in >> 27) & 0x1)<< 2)|(((GPIO.in >> 23) & 0x1)<< 3)|(((GPIO.in >> 14) & 0x1)<< 4)|(((GPIO.in >> 15) & 0x1)<< 5)|(((GPIO.in >> 16) & 0x1)<< 6)|(((GPIO.in >> 21) & 0x1)<< 7)|(((GPIO.in >> 33) & 0x1)<< 8)|(((GPIO.in >> 2) & 0x1)<< 9)|(((GPIO.in >> 4) & 0x1)<< 10)|(((GPIO.in >> 12) & 0x1)<< 11)|(((GPIO.in >> 22) & 0x1)<< 12)|(((GPIO.in >> 19) & 0x1)<< 13)|(0<<14)|(0<<15));
      uint8_t a[] = {resulte,(resulte>>8)};

      REPORT->setValue(a, sizeof(a));
      REPORT->notify();
  }
}

CybershoesVR, 2022-10-17:
same here. On PC reconnect works, but on Quest2 does not work.

h2zero, 2022-10-17:
Can either of you share a debug log output when this occurs?

JuanJoseMoralesCalvo, 2022-10-17:
The workaround for me was to add security to the BLE, but honestly i dont know the reason :D

Attach to setup() after advertising:

  NimBLESecurity *SECURITY_MANDATORY = new BLESecurity();
  SECURITY_MANDATORY->setAuthenticationMode(ESP_LE_AUTH_BOND);

CybershoesVR, 2022-10-18:
Thank you but this did not work. I tried a few other setting but decided it'll be better to go with debugging as @h2zero suggested.

from the log: 
**GamePadPlus V3** is a BT gamepad device that reconnects successful.
**Cybershoes2** is our BT gamepad device that does reconnect but gamepad functionality fails upon reconnect.

Dear  @h2zero, please can you hint at what makes the difference?

[odh_logs_2022-10-18 13.55.53.833.txt](https://github.com/h2zero/NimBLE-Arduino/files/9811161/odh_logs_2022-10-18.13.55.53.833.txt)

h2zero, 2022-10-18:
Looks like you need to enable bonding. Try adding this after initialization:
`NimBLEDevice::setSecurityAuth(true, true, true);'

Have a look at the secure client/server examples.

CybershoesVR, 2022-10-18:
yes, I did that as you see below.
I am using NimBLE 1.38 with ESP32-BLE-Gamepad 0.5.1, because using NimBLE 1.40 makes my ESP32-C crash.

```
void BleGamepad::taskServer(void *pvParameter)
{
    BleGamepad *BleGamepadInstance = (BleGamepad *)pvParameter; // static_cast<BleGamepad *>(pvParameter);
    NimBLEDevice::init(BleGamepadInstance->deviceName);
    NimBLEServer *pServer = NimBLEDevice::createServer();
    pServer->setCallbacks(BleGamepadInstance->connectionStatus);

    BleGamepadInstance->hid = new NimBLEHIDDevice(pServer);
    BleGamepadInstance->inputGamepad = BleGamepadInstance->hid->inputReport(BleGamepadInstance->configuration.getHidReportId()); // <-- input REPORTID from report map
    BleGamepadInstance->connectionStatus->inputGamepad = BleGamepadInstance->inputGamepad;

    BleGamepadInstance->hid->manufacturer()->setValue(BleGamepadInstance->deviceManufacturer);

    BleGamepadInstance->hid->pnp(0x01, vid, pid, 0x0110);
    BleGamepadInstance->hid->hidInfo(0x00, 0x01);

    NimBLESecurity *SECURITY_MANDATORY = new BLESecurity();
    SECURITY_MANDATORY->setAuthenticationMode(ESP_LE_AUTH_BOND); // ESP_LE_AUTH_REQ_BOND_MITM,
    SECURITY_MANDATORY->setCapability(ESP_IO_CAP_NONE);

    /**
     * @brief Set the authorization mode for this device.
     * @param bonding If true we allow bonding, false no bonding will be performed.
     * @param mitm If true we are capable of man in the middle protection, false if not.
     * @param sc If true we will perform secure connection pairing, false we will use legacy pairing.
     */
    
    //NimBLEDevice::setSecurityAuth(bool bonding, bool mitm, bool sc);
    NimBLEDevice::setSecurityAuth(true, true, true);
    
    //NimBLEDevice::setSecurityAuth( BLE_SM_PAIR_AUTHREQ_MITM); // BEST must pair again each time but no need to unpair
    //NimBLEDevice::setSecurityAuth(BLE_SM_PAIR_AUTHREQ_SC);    //same as usual

    uint8_t *customHidReportDescriptor = new uint8_t[hidReportDescriptorSize];
    memcpy(customHidReportDescriptor, tempHidReportDescriptor, hidReportDescriptorSize);

    for (int i = 0; i < hidReportDescriptorSize; i++)
        Serial.printf("%02x", customHidReportDescriptor[i]);

    BleGamepadInstance->hid->reportMap((uint8_t *)customHidReportDescriptor, hidReportDescriptorSize);
    BleGamepadInstance->hid->startServices();

    BleGamepadInstance->onStarted(pServer);

    NimBLEAdvertising *pAdvertising = pServer->getAdvertising();
    pAdvertising->setAppearance(HID_GAMEPAD);
    pAdvertising->addServiceUUID(BleGamepadInstance->hid->hidService()->getUUID());
    pAdvertising->start();
    //https://github.com/h2zero/NimBLE-Arduino/issues/464
    //The workaround for me was to add security to the BLE, but honestly i dont know the reason :D
    //Attach to setup() after advertising:
    //NimBLESecurity *SECURITY_MANDATORY = new BLESecurity();
    //SECURITY_MANDATORY->setAuthenticationMode(ESP_LE_AUTH_BOND);
    //end of inserted code

    BleGamepadInstance->hid->setBatteryLevel(BleGamepadInstance->batteryLevel);

    ESP_LOGD(LOG_TAG, "Advertising started!");
    vTaskDelay(portMAX_DELAY); // delay(portMAX_DELAY);
}
```

 In a variation I also tried without 
```
    NimBLESecurity *SECURITY_MANDATORY = new BLESecurity();
    SECURITY_MANDATORY->setAuthenticationMode(ESP_LE_AUTH_BOND); // ESP_LE_AUTH_REQ_BOND_MITM,
    SECURITY_MANDATORY->setCapability(ESP_IO_CAP_NONE);
```

and also tried to move 
```
NimBLEDevice::setSecurityAuth(true, true, true);
```
after pAdvertising->start();

CybershoesVR, 2022-10-18:
reconnecting our Cybershoes gives an **encryption failed**:
```
15:55:28.310  1683  1704 D BluetoothGattServer: onConnectionUpdated() - Device=34:B4:72:00:A1:D6 interval=9 latency=0 timeout=600 status=0
15:55:28.356  1412  2092 W bt_btif       : btif_gatt_set_encryption_cb() - Encryption failed (1)
15:55:28.356  1412  2092 E bt_btif       : bta_hh_security_cmpl() - encryption failed; status=0x000e, reason=0x000a
```

reconnecting GamePadPlus
```
15:54:46.249  1683  1704 D BluetoothGattServer: onConnectionUpdated() - Device=20:21:03:07:B9:98 interval=12 latency=0 timeout=200 status=0
15:54:46.331  1412  2092 W bt_stack      : [WARNING:bta_gattc_api.cc(652)] notification already registered
15:54:46.332  1412  2092 I chatty        : uid=1002(bluetooth) bt_main_thread identical 1 line
15:54:46.332  1412  2092 W bt_stack      : [WARNING:bta_gattc_api.cc(652)] notification already registered
15:54:46.332  1412  2092 W bt_btif       : bta_hh_co_open: Found an existing device with the same handle dev_status=2, address=20:21:03:07:b9:98, attr_mask=0x0000, sub_class=0x00, app_id=255
15:54:46.332  1412  1662 W bt_btif       : btif_hh_upstreams_evt: BTA_HH_OPN_EVT: handle=16, status =0
15:54:46.332  1412  1662 W bt_btif       : BTA_HH_OPEN_EVT: Found device...Getting dscp info for handle ... 16
15:54:46.332  1412  1662 I bt_btif_dm    : get_cod remote_cod = 0x00000504
15:54:46.332  1412  1662 I bt_btif_dm    : get_cod remote_cod = 0x00000504
15:54:46.336  1412  1412 D BluetoothAdapterService: getAdapterService() - returning com.android.bluetooth.btservice.AdapterService@590f4fe
15:54:46.336  1412  1662 W bt_btif       : btif_hh_upstreams_evt: name = GamePadPlus V3
15:54:46.336  1412  1662 W bt_btif       : bta_hh_co_send_hid_info: fd = 108, name = [GamePadPlus V3], dscp_len = 187
15:54:46.336  1412  1662 W bt_btif       : bta_hh_co_send_hid_info: vendor_id = 0x1949, product_id = 0x0402, version= 0x0111,ctry_code=0x00
15:54:46.347  1412  1412 D BluetoothAdapterService: isQuetModeEnabled() - Enabled = false
15:54:46.350  1412  1662 W bt_btif       : bta_hh_co_send_hid_info: wrote descriptor to fd = 108, dscp_len = 187, result = 0
15:54:46.350  1412  1662 W bt_stack      : [WARNING:btif_hh.cc(430)]  Device 20:21:03:07:b9:98 already added
15:54:46.350  1412  1662 W bt_btif       : btif_hh_upstreams_evt: Device already added 
15:54:46.359  1412  1412 D BluetoothAdapterService: getAdapterService() - returning com.android.bluetooth.btservice.AdapterService@590f4fe
15:54:46.359  1412  1412 V BluetoothDatabase: getProfilePriority: 20:21:03:07:B9:98, profile=4, priority = 100
15:54:46.360  1412  1412 D BluetoothHidHostService: Connection state 20:21:03:07:B9:98: 0->2
15:54:46.361  1753  1753 D CachedBluetoothDevice: onProfileStateChanged: profile HID, device=20:21:03:07:B9:98, newProfileState 2
15:54:46.361  1412  1412 D AdapterProperties: PROFILE_CONNECTION_STATE_CHANGE: profile=4, device=20:21:03:07:B9:98, 0 -> 2
15:54:46.362  1412  1412 W AdapterProperties: PROFILE_CONNECTION_STATE_CHANGE: unexpected transition for profile=4, device=20:21:03:07:B9:98, 0 -> 2
15:54:46.363  1412  1412 D AdapterProperties: ADAPTER_CONNECTION_STATE_CHANGE: 20:21:03:07:B9:98: 0 -> 2
15:54:46.363  1412  1412 W AdapterProperties: ADAPTER_CONNECTION_STATE_CHANGE: unexpected transition for profile=4, device=20:21:03:07:B9:98, 0 -> 2
15:54:46.365  1683  1683 I BleModule     : BluetoothAdapter: Device[20:21:03:07:B9:98] Connection State changed from 0 to 0
```

and finally 
```
15:54:46.547  1034  1142 I InputReader   : Device added: id=15, name='GamePadPlus V3', sources=0x01000511
15:54:46.548  2020  2020 D GamepadListener: Adding Device: GamePadPlus V3 pid: 1026 sources: 16778513
15:54:46.549  3026  3026 D GamepadListener: Adding Device: GamePadPlus V3 pid: 1026 sources: 16778513
15:54:46.549  2236  2236 I [OAO] ShellOverlayServiceJNI: NativePTKServiceNotifyDeviceConnectionChanged
15:54:46.549  2236  2236 D [OAO] ShellApplication: PTKService enumerated keyboard: {vendorId: 0x1949 (0n6473), productId: 0x402 (0n1026)}
15:54:46.554  1734  1734 D GamepadListener: Adding Device: GamePadPlus V3 pid: 1026 sources: 16778513
15:54:46.577  2971  2971 D GamepadListener: Adding Device: GamePadPlus V3 pid: 1026 sources: 16778513
```

CybershoesVR, 2022-10-18:
also no success with:
```
    NimBLEDevice::setSecurityAuth(true, true, true);
    NimBLESecurity *SECURITY_MANDATORY = new BLESecurity();

    /**
     * @brief Set requested authentication mode
     * @param [in] auth_req A bitmask containing one or more of:
     * * ESP_LE_AUTH_NO_BOND              0x00
     * * ESP_LE_AUTH_BOND                 0x01
     * * ESP_LE_AUTH_REQ_MITM             (1 << 2)
     * * ESP_LE_AUTH_REQ_BOND_MITM        (ESP_LE_AUTH_BOND | ESP_LE_AUTH_REQ_MITM)
     * * ESP_LE_AUTH_REQ_SC_ONLY          (1 << 3)
     * * ESP_LE_AUTH_REQ_SC_BOND          (ESP_LE_AUTH_BOND | ESP_LE_AUTH_REQ_SC_ONLY)
     * * ESP_LE_AUTH_REQ_SC_MITM          (ESP_LE_AUTH_REQ_MITM | ESP_LE_AUTH_REQ_SC_ONLY)
     * * ESP_LE_AUTH_REQ_SC_MITM_BOND     (ESP_LE_AUTH_REQ_MITM | ESP_LE_AUTH_REQ_SC_ONLY | ESP_LE_AUTH_BOND)
     */
    SECURITY_MANDATORY->setAuthenticationMode(ESP_LE_AUTH_REQ_BOND_MITM); // ESP_LE_AUTH_REQ_BOND_MITM,ESP_LE_AUTH_BOND
    //SECURITY_MANDATORY->setCapability(ESP_IO_CAP_NONE);

    SECURITY_MANDATORY->setInitEncryptionKey(BLE_SM_PAIR_KEY_DIST_ENC);
    SECURITY_MANDATORY->setRespEncryptionKey(BLE_SM_PAIR_KEY_DIST_ENC);
```

CybershoesVR, 2022-10-18:
sometimes (rareley) reconnection works succesfully with:
```
NimBLEDevice::setSecurityAuth(true, true, true);
```

h2zero, 2022-10-18:
I'm not too sure what the error is yet, the logs suggest an error in a value passed during pairing. What kind of pairing does the other device expect? 

If not using PIN codes or any other mitm protection then try setting the last 2 parameters of the `NimBLEDevice::setSecurityAuth` call to false. Otherwise just set the last parameter to false if mitm is required.

h2zero, 2022-10-18:
You should also remove this:

```

NimBLESecurity *SECURITY_MANDATORY = new BLESecurity();

    /**
     * @brief Set requested authentication mode
     * @param [in] auth_req A bitmask containing one or more of:
     * * ESP_LE_AUTH_NO_BOND              0x00
     * * ESP_LE_AUTH_BOND                 0x01
     * * ESP_LE_AUTH_REQ_MITM             (1 << 2)
     * * ESP_LE_AUTH_REQ_BOND_MITM        (ESP_LE_AUTH_BOND | ESP_LE_AUTH_REQ_MITM)
     * * ESP_LE_AUTH_REQ_SC_ONLY          (1 << 3)
     * * ESP_LE_AUTH_REQ_SC_BOND          (ESP_LE_AUTH_BOND | ESP_LE_AUTH_REQ_SC_ONLY)
     * * ESP_LE_AUTH_REQ_SC_MITM          (ESP_LE_AUTH_REQ_MITM | ESP_LE_AUTH_REQ_SC_ONLY)
     * * ESP_LE_AUTH_REQ_SC_MITM_BOND     (ESP_LE_AUTH_REQ_MITM | ESP_LE_AUTH_REQ_SC_ONLY | ESP_LE_AUTH_BOND)
     */
    SECURITY_MANDATORY->setAuthenticationMode(ESP_LE_AUTH_REQ_BOND_MITM); // ESP_LE_AUTH_REQ_BOND_MITM,ESP_LE_AUTH_BOND
    //SECURITY_MANDATORY->setCapability(ESP_IO_CAP_NONE);

    SECURITY_MANDATORY->setInitEncryptionKey(BLE_SM_PAIR_KEY_DIST_ENC);
    SECURITY_MANDATORY->setRespEncryptionKey(BLE_SM_PAIR_KEY_DIST_ENC);

```

That is conflicting with the security settings above it.

JuanJoseMoralesCalvo, 2022-10-19:
@CybershoesVR Try this workaround from chewara, it sounds like its happening the same:

class MyCallbacks : public BLEServerCallbacks {
  void onConnect(BLEServer* pServer){
    Serial.println("connected");
    BLEDescriptor *desc = input->getDescriptorByUUID(BLEUUID((uint16_t)0x2902));
    uint8_t val[] = {0x01, 0x00};
    desc->setValue(val, 2);
  }

  void onDisconnect(BLEServer* pServer){
        Serial.println("disconnected");
  }
};

Issue Open:
https://github.com/nkolban/esp32-snippets/issues/632

JuanJoseMoralesCalvo, 2022-10-19:
You have to add something like this on the onConenct:

BLEDescriptor *desc = input->getDescriptorByUUID(BLEUUID((uint16_t)0x2902));
uint8_t val[] = {0x01, 0x00};
desc->setValue(val, 2);

fabdelgado, 2022-11-05:
I’m facing the same issue. 

@JuanJoseMoralesCalvo Did your suggestion work for you?

JuanJoseMoralesCalvo, 2022-11-05:
The workaround on comment #12 didnt work as i understand its only for the nkolban esp32 libraries. Finally by adding security it worked.

NimBLESecurity *SECURITY_MANDATORY = new NimBLESecurity();
SECURITY_MANDATORY->setAuthenticationMode(ESP_LE_AUTH_BOND); // ESP_LE_AUTH_REQ_BOND_MITM,
SECURITY_MANDATORY->setCapability(ESP_IO_CAP_NONE);

fabdelgado, 2022-11-05:
On which lines of which file should I put this?  

Please explain me if I should add or replace code?

h2zero, 2022-11-05:
I would encourage not to use the security class as it has need deprecated.

The equivalent call is simply:
```
NimBLEDevice::setSecurityAuth(true, false, false); // enable bonding, no MITM, no SC

```

The IO caps default to NONE. Just put that line somewhere after initializing NimBLE.

fabdelgado, 2022-11-05:
@h2zero like this example in my gist?

https://gist.github.com/fabdelgado/60c715dec2651b46a0d342c46179fe48

h2zero, 2022-11-05:
Looks good to me

fabdelgado, 2022-11-05:
I will try that!

fabdelgado, 2022-11-06:
It has not worked for me with my chromecast v2 Google TV.
You can review the video I have recorded for you.

https://user-images.githubusercontent.com/8621547/200147816-289740df-03e3-4b02-9f8e-d16e6ccdcda3.mp4

h2zero, 2022-11-06:
Hmm, please try erasing the flash of the esp32 then upload your code. Also remove any devices bonded to the TV if you can.

fabdelgado, 2022-11-06:
@h2zero this worked for me, thanks for the help.

https://user-images.githubusercontent.com/8621547/200183433-a0bfaea0-bfe1-4ad7-8a33-ad92f576229e.mp4

This is the code that I have used, I have left it for future issues.

`void setup() {
NimBLEDevice::init("NimBLE");
NimBLEDevice::setSecurityAuth(true, false, false); // enable bonding, no MITM, no SC
}`

fabdelgado, 2022-11-06:
@h2zero I have another related problem, when I press forget device and then restart the Chromecast it unbinds the remote from Chromecast instead of ESP32.
This problem happened to me before the problem of connecting and disconnecting.
Do you have any idea?

h2zero, 2022-11-07:
I don't have any idea why that would happen, wouldn't have anything to do with the esp32 though.

closed 2024-06-04 by h2zero