h2zero/NimBLE-Arduino #465: Getting different MAC address of the same device.

HeavyShmel, 2022-10-20:
I'm trying to make a dynamic whitelist on esp32.

To do it, I store the addresses of devices with which I had a connection in NVS. If esp32 have a reboot, it retrieve addresses from the NVS and add them to the Whitelist.

But I ran into a problem. The smartphone from which I connect to the device almost every time has a different MAC address. And I get the same problem with other smarphone. And I have never got a real MAC address of my smartphone on esp32. So what could be a reason?

I use this server callback in which I get an address of smartphone:

void ServerCallbacks::onConnect(NimBLEServer *pServer, ble_gap_conn_desc *desc)
{
    NimBLEAddress tmp(desc->peer_ota_addr);
    if(!NimBLEDevice::onWhiteList(tmp)){
        NimBLEDevice::whiteListAdd(tmp);
        addAddressToNVS(tmp);                     //my function in whitch i save an address
    }
    pAdvertising->stop();
}

sanastasiou, 2023-06-07:
Sorry to reply to this very old thread, but I seem to have the exact same issue. 

Although my device is bonding, how would I get the static address? According to https://github.com/h2zero/NimBLE-Arduino/blob/release/1.4/examples/NimBLE_Server_Whitelist/NimBLE_Server_Whitelist.ino

That would be `desc->peer_ota_addr` but that is not correct, since that address is always changing.

h2zero, 2024-06-04:
the `desc->peer_id_addr` would be what is needed. But this has changed going forward.

closed 2024-06-04 by h2zero