h2zero/NimBLE-Arduino #467: Possible to avoid polling loop for new connections?

marchingband, 2022-10-21:
I have an application where heap memory is very low. I call `vTaskDelete(NULL)` in the Arduino `loop()` task to save heap, and would like to use your library **without** having to create a new RTOS task, which consumes heap, just to poll for new connections, but all your examples that I could find poll inside the `loop()` function, and run the connection logic there. Using this method in a new FreeRTOS task works great, it's an amazing library, but I would like to avoid creating the task if at all possible.

I have tried to implement the connection inside the `AdvertisedDeviceCallbacks::onResult()` callback but the connection always fails with a timeout. I am unclear why that is ... would a `delay` in here potentially work better? Is there another callback where I could put the connection logic that would work better? Any other tips for me? Thank you so much!!!!

```
class AdvertisedDeviceCallbacks: public NimBLEAdvertisedDeviceCallbacks {
    void onResult(NimBLEAdvertisedDevice* advertisedDevice) {
        log_i("found device %s", advertisedDevice->toString().c_str());
        if(advertisedDevice->isAdvertisingService(NimBLEUUID(SERVICE_UUID)))
        {
            log_i("it's midi");
            NimBLEDevice::getScan()->stop();
            NimBLEClient* pClient = NimBLEDevice::createClient();
            pClient->setClientCallbacks(&clientCB, false);
            pClient->setConnectionParams(6, 6, 0, 200);
            res = pClient->connect(advertisedDevice);
            if(!res){log_e("connect failed");}

            NimBLERemoteService* pSvc = pClient->getService(SERVICE_UUID);            
            if(!pSvc){log_e("get service failed");}

            NimBLERemoteCharacteristic* pChr = pSvc->getCharacteristic(CHARACTERISTIC_UUID);
            if(!pChr){log_e("get char failed");}

            res = pChr->subscribe(true, notifyCB);
            if(!res){log_e("subscribe failed");}
        }
    };
};
```

h2zero, 2022-10-21:
The reason calling connect fails in the callback is because it is a blocking function and doing so there blocks the BLE stack entirely. I've considered ways to change this to avoid blocking but so far I haven't been happy with the results.

What I've done lately is use a queue or task notification and set an event in the callback to trigger a connection. This still requires a task that can be blocked though. You can minimize the task stack by checking the high water mark and reduce the size as needed. This will certainly be less than the loop task.

How much heap space do you need to free up?

marchingband, 2022-10-22:
I got stack overflows with a 2048 byte stack, but it seems to be stable with a 3072 byte stack.
I'm surprised the BLE stack doesn't provide the memory needed in some way, but thats a bit beyond my knowledge.
Ideally I would like to not have to allocate at all.
I found 3k elsewhere to spare, so the app is working now, but let me know if you find a way to improve this, and thanks for getting back :)

h2zero, 2022-10-24:
That's just the rock n roll, I'll certainly look to improve this anywhere possible.

closed 2022-10-24 by h2zero