h2zero/NimBLE-Arduino #468: ESP32 Devkit keep rebooting for NimBLE latest version

mrezzaaa, 2022-10-23:
Hi, im trying to create device with multiple function both WiFi and BLE (dual mode). 

When esp sending to another nearby esp via BLE characteristics its success but after that crashing and causing bootloop after 30-60second booted up.

Tracing some debug, here what i got. 

`D NimBLEClient: ">> disconnect()"
E NimBLEClient: "ble_gap_terminate failed: rc=530 "
D NimBLEClient: "<< disconnect()"
D NimBLEClient: ">> connect(94:b9:7e:d9:34:b2)"
E NimBLEClient: "Client busy, connected to 94:b9:7e:d8:d7:02, id=0"
D NimBLEClient: ">> disconnect()"
D NimBLEClient: "<< disconnect()"
D NimBLEClient: "Got Client event "
D NimBLEClientCallbacks: "onDisconnect: default"
D NimBLEClient: ">> deleteServices"
D NimBLERemoteService: ">> deleteCharacteristics"
D NimBLERemoteCharacteristic: ">> deleteDescriptors"
D NimBLERemoteCharacteristic: "<< deleteDescriptors"
D NimBLERemoteService: "<< deleteCharacteristics"
D NimBLEClient: "<< deleteServices"
Command sent!
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400e7164  PS      : 0x00060830  A0      : 0x800d6935  A1      : 0x3ffcf660  
A2      : 0x3ffc3400  A3      : 0x3f40082c  A4      : 0x3f40201b  A5      : 0x3ffcf630  
A6      : 0x00000003  A7      : 0x00060223  A8      : 0x800e7164  A9      : 0x3ffcf630  
A10     : 0x00000000  A11     : 0x00007530  A12     : 0x3f40201b  A13     : 0x00000000  
A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000`

Also already modifying config in `idf.py menuconfig` Try to moving ble run on core 1 still bootloop. 
Trying just only server also bootloop.
I have no idea what's going on :(

h2zero, 2022-10-23:
Looks like a nullptr exception, what is called when `Command sent!` is output? I suspect whatever that is will likely be NULL and should be tested for that beforehand.

mrezzaaa, 2022-10-23:
it's just stat after sending data `lightgroup:SI0isThQkVdkI4r2qU0U:43` to nearby devices in std::string .
it's also happen when i trying to create simple BLE server with 2 services and 5 characteristics in total.
Just starting server and no any complex code .
FYI i use idf version `ESP-IDF v4.3.1-dirty`
with config in ble :

- BLE only mode
- Default stack size  (idf default config 4096)
- 3 max connection
- Tried to run on code 0 or 1 to find out bootloop happen again ( and its happen )
- Bluetooth host NimBLE only

is needed to enable `blufi` options to get BLE and WIFI work ?

mrezzaaa, 2022-10-23:
Hi, i fixed the issue by updating esp idf version to 4.4.0 and also arduino-espressif to 4.4.0 
Now they run stable .

closed 2022-10-23 by mrezzaaa

h2zero, 2022-10-24:
Thanks for the update!