h2zero/NimBLE-Arduino #474: Server mode send a lot of bytes in buffer alone

rafaelpm, 2022-11-17:
Everyone know about lack components, so I need to replace a Bluetooth NINA-B112.
I started a project from example server, but editing lines like bellow:
```
NimBLEDevice::init("NINA-B112");
NimBLEDevice::setSecurityAuth(false,false,false);
pSerialService = pServer->createService(UUID_SERVICE_SERIAL_PORT);
pSerialCharacteristic = pSerialService->createCharacteristic(
			  UUID_CHARACTERISTIC_FIFO,
			  NIMBLE_PROPERTY::READ |
			  NIMBLE_PROPERTY::WRITE |
			  NIMBLE_PROPERTY::NOTIFY 
			);
pSerialCharacteristic->setValue("RAFAEL");
pSerialCharacteristic->notify(true);
pSerialCharacteristic->setCallbacks(&chrCallbacks);
pSerialService->start();

pAdvertising = NimBLEDevice::getAdvertising();
pAdvertising->addServiceUUID(pSerialService->getUUID());
pAdvertising->setScanResponse(true);
pAdvertising->start();
```
My defines are:
```
#define UUID_SERVICE_SERIAL_PORT "2456e1b9-26e2-8f83-e744-f34f01e9d701"
#define UUID_CHARACTERISTIC_FIFO "2456e1b9-26e2-8f83-e744-f34f01e9d703"
```
I used u-blox app: https://play.google.com/store/apps/details?id=com.ublox.BLE
Using this android app, when connected, I receive a lot of same data like: RAFAEL.
If I send data in chat tab (for example: Test), I receive a lot of "Test".
My conclusion is that it sends what is in the buffer, but what started it?
How control it?

h2zero, 2022-11-18:
When you send notifications it sends the current value of the characteristic, if you're doing this in a loop it will keep sending the notification.

rafaelpm, 2022-11-18:
Removing NIMBLE_PROPERTY::NOTIFY so don't send in a loop, however, don't send anything.
How set and reset notify flag in runtime? Is need to stop and start server?

rafaelpm, 2022-11-18:
Thanks for the reply, bug has been fixed. 
I wrote the below code in my loop.
```
if(pServer->getConnectedCount() > 0) {
  NimBLEService* pSvc = pServer->getServiceByUUID(UUID_SERVICE_SERIAL_PORT);
  if(pSvc) {	  
	  NimBLECharacteristic* pChr = pSvc->getCharacteristic(UUID_CHARACTERISTIC_FIFO);
	  if(pChr) {		  
		  pChr->notify(false);		  
	  }
  }
}
```

closed 2022-11-18 by rafaelpm