h2zero/NimBLE-Arduino #477: Can't get descriptor of a service

Jimbo210, 2022-11-21:
Hi, opening this issue as i wasn't able to find any solution by myself or through the web. 
I'm kind of new to the world of ble and esp32. My final goal is to move a code I got running on pc to the esp32, but since I used different libraries I still have to aclimatize to the new set of functions. 

I'm currently stuck at trying to get a descriptor of a service, to subsequently send some bytes and subscribe for notification ecc... 

From my understanding the hierarchy of the library is: Client -> Service -> Characterstic -> Descriptor and each step has it's own set of functions, hence I'm not able to call getDescriptors() from a service. 

Am I missing something or can it be done? 

(As a spot of context I'll leave the screenshot of nRF Connect to show what I'm trying to see/get with the esp32).
[image]

Btw if I call getDescriptors() from either of the two Characteristics I get a vector of size 0, so that's not the way...

Thank in advance for the replies.
Best Regards.

h2zero, 2022-11-22:
As you can see in the nRFConnect screenshot, the characteristic `NimBLEUUID charUuid1("8b00ace7-eb0b-49b0-bbe9-9aee0a26e1a3");` does not have notification capabilities.

You are attempting to subscribe to it here:
```

        NimBLERemoteCharacteristic* pCharacteristic1 = pService->getCharacteristic(charUuid1);
        
        if(pCharacteristic1->subscribe(true, notificationCallBack, true)){
          Serial.println("Subscribed");
        }

        if(pCharacteristic1->canNotify()){
          Serial.println("Can Notify");
        }
``` 
`canNotify()` should return false here for this characteristic.

I believe you meant to use the `charUuid2` here, which as shown, has notification property enabled.

Jimbo210, 2022-11-22:
First of all thanks for the reply and for making me check again.

<charUuid2> was in fact the one which I should have been subscribing to and I got everything working.

I'll point out some issues that I have found, even though I later realized I didn't actually need those functions results (but it might be helpful to someone or I could learn why they were there).

The descriptor which I was looking for (that I didn't need) was in deed "within"  <charUuid2>, but getting to it wasn't so easy:
 - pCharacteristic->getDescriptors(); returned a vector of size 0...
 - if I did the same thing after subscribing to the characteristic I get a vector of size 1 and could subsequently get to it's "features".
 - Couldn't get pCharacteristic->getDescritptor(UuidDescriptor) working tough...

Current (small) issue I have is I get two responses from the notificationCallBack function as if it's called twice:

Code: 
<

#include <NimBLEDevice.h>

  static NimBLEClient* pClient;

  NimBLEUUID serviceUuid("14839ac4-7d7e-415c-9a42-167340cf2339");
  NimBLEUUID charUuid1("8b00ace7-eb0b-49b0-bbe9-9aee0a26e1a3");
  NimBLEUUID charUuid2("0734594a-a8e7-4b1a-a6b1-cd5243059a57");

  std::vector<uint8_t> wrBv = { 0xAA, 0x17, 0xE8, 0x00, 0x00, 0x00, 0x00, 0x1B };

  NimBLERemoteCharacteristic* pChr1;

void whatToDo(NimBLERemoteCharacteristic* pRemoteCharacteristic, uint8_t pData[], size_t length, bool isNotify){
  Serial.println("Been here");
}

void setup() {

  Serial.begin(115200);

  Serial.println("Starting NimBLE Client");
  NimBLEDevice::init("");

  pClient = NimBLEDevice::createClient(NimBLEAddress("ce:1a:74:6b:2d:bc", 1));
  Serial.println("New client created");

  if(pClient->connect()) {
    Serial.println("Connected to device.");

    NimBLERemoteService* pService = pClient->getService(serviceUuid);

    NimBLERemoteCharacteristic* pCharacteristic1 = pService->getCharacteristic(charUuid1);
    NimBLERemoteCharacteristic* pCharacteristic2 = pService->getCharacteristic(charUuid2);

      if(pCharacteristic2->canNotify()){
        Serial.println("Can notify.");
        if(pCharacteristic2->subscribe(true, whatToDo, true)){
          Serial.println("subscibed.");
          pChr1 = pCharacteristic1;
        }
      }
  }
}

void loop() {
  pChr1->writeValue(wrBv);
  delay(3000);
}

>

Serial monitor:

<
19:09:36.505 -> Starting NimBLE Client
19:09:36.759 -> New client created
19:09:37.829 -> Connected to device.
19:09:38.545 -> Can Notify.
19:09:38.837 -> Subscribed
19:09:38.955 -> Been here
19:09:38.955 -> Been here
19:09:41.956 -> Been here
19:09:41.956 -> Been here
19:09:44.881 -> Been here
19:09:44.881 -> Been here
>

Thanks again for the reply and library

h2zero, 2022-11-23:
> pCharacteristic2->getDescriptors(); returned a vector of size 0...

This occurs if you have not retrieved the descriptors yet.

> if I did the same thing after subscribing to the characteristic I get a vector of size 1 and could subsequently get to it's "features".

The call to subscribe will retrieve the descriptor info from the server so this is normal.

> Couldn't get pCharacteristic2->getDescritptor(UuidDescriptor) working though

What were you using for the UUID here? Not sure why you need to get the CCCD, could you explain a bit?

> Current (small) issue I have is I get two responses from the notificationCallBack function as if it's called twice:

That's strange, maybe the device sends the notification twice?

closed 2022-12-02 by Jimbo210