h2zero/NimBLE-Arduino #479: NimBLE-Arduino uses non a 'typical' pairing PIN 123456

IhorNehrutsa, 2022-11-24:
https://github.com/h2zero/NimBLE-Arduino/blob/0d9f039ba7f960946a8c9aeb639c3a64294f0d60/src/NimBLEDevice.cpp#L69
https://github.com/h2zero/NimBLE-Arduino/blob/0d9f039ba7f960946a8c9aeb639c3a64294f0d60/src/NimBLEClient.cpp#L1278-L1281
https://github.com/h2zero/NimBLE-Arduino/blob/0d9f039ba7f960946a8c9aeb639c3a64294f0d60/src/NimBLEServer.cpp#L899-L902
But Android proposes 0000 or 1234 as default
[image]

chegewara, 2022-11-24:
This is some old app, or maybe misleading users.
Bluetooth specs says that pairing pin has to be 6 digits, no more and not less, always 6. Maybe that app is adding leading `0`s.

IhorNehrutsa, 2022-11-24:
Android 7.0
Phone: Meizu M5 Note Model:M621H

IhorNehrutsa, 2022-11-24:
@chegewara  
>This is some old app

Meshtastic app uses 123456 PIN code
```
// Default Bluetooth PIN
#define defaultBLEPin 123456
```
https://github.com/meshtastic/firmware/blob/c87cd136d4398d69829327e28145ee20a845f43d/src/DebugConfiguration.h#L38

It seems that it is not an app message, but the Android message. Android 12 proposes the same:

Android 12
Samsung Galaxy S20 FE 5G
[image]

chegewara, 2022-11-24:
It does not change the fact you have to input `001234`.

h2zero, 2022-11-25:
The dialog in the screenshot is showing a suggestion for classic bluetooth. As @chegewara rightly pointed out, in BLE the spec is a 6 digit pin, you should still be able to enter `123456` and pair with it.

IhorNehrutsa, 2022-11-25:
@chegewara @h2zero Thanks.

closed 2022-11-25 by IhorNehrutsa