h2zero/NimBLE-Arduino #481: When scanning, callback method not called for some advertisements received, causes memory leak

Bascy, 2022-11-25:
We are running a scanner indefinitely on our ESP32, with duplicates enabled and maxResult == 0.
This results in continues decrease of available memory.

W are getting a lot of advertisements (vrom for me unknown devices) which are added to the scanResult m_advertisedDevicesVector but for which the callback method is never called, and hence they are never removed from the vector.

To test this I added some code to periodically remove all advertisements in the vector that are older than 10 sec, and now my memory use is stable.

I can't find any documentation on the event_type's that are used in if statements in NimBKEScan.cpp (around line 143) so I don;t know what the root cause of this is. Why is the onResult() method for these advertisements never called, but for other advertisements the callback is called?

Hope anyone can shed some light on this

h2zero, 2022-11-25:
This is caused by active scanning. If a scan response is never received the device will remain in the buffer until the response is received and the callback called. Try testing with active scanning disabled and let me know.

Bascy, 2022-11-25:
thanks. this seemed to have solve the problem.  now let's hope that now of
the devices we use need active scanning ...

closed 2022-11-25 by Bascy