h2zero/NimBLE-Arduino #482: LOG_LEVEL defines collision

vovagorodok, 2022-11-26:
At `log_common.h` we have defined:
```
#define LOG_LEVEL_DEBUG    (0)
#define LOG_LEVEL_INFO     (1)
#define LOG_LEVEL_WARN     (2)
#define LOG_LEVEL_ERROR    (3)
#define LOG_LEVEL_CRITICAL (4)
#define LOG_LEVEL_NONE     (5)
```
`LOG_LEVEL..` names are very common and I have redefinition issue when another than NibBle logger is used.\
Because its internal NibBle logger, proposed to use some internal names like:
`BLE_LOG_LEVEL..` or `NIM_BLE_LOG_LEVEL..`

h2zero, 2022-11-30:
This file is maintained [upstream](https://github.com/espressif/esp-nimble/blob/nimble-1.4.0-idf/porting/nimble/include/log_common/log_common.h), this issue would be better to bring up there.

vovagorodok, 2022-11-30:
Created. Lets wait the answer:
https://github.com/espressif/esp-nimble/issues/49

h2zero, 2024-06-04:
This will be resolved when updated by espressif.