h2zero/NimBLE-Arduino #484: Peer removed pairing information

foofel, 2022-11-30:
I have a device with mostly public but some private data. If too many devices bond, it seems like the pairing information does not get stored any more (ios returns Peer removed pairing information). After some googling i found that it seems only 3 bounds can be stored by default, that is fine with me as normally never more than one device needs to be bonded at a time, but it looks like this is the overall limit. Would it be possible to implement some "rolling" bonding where a new bond kicks out the oldest one if the list is full?

h2zero, 2022-11-30:
This is the behavior that should already be happening. When the bond limit is reached the oldest one gets deleted and the new bond is created. If this is not working then there could be a bug.

foofel, 2022-11-30:
Hmm, then it seems like its another problem, ill take a further look. What would be the best way to debug this? What debug messages should i look for?

CharlesXSong, 2023-03-31:
Oh, seems like we met the same problem: [https://github.com/h2zero/NimBLE-Arduino/issues/522](url)

h2zero, 2024-06-04:
Closing as stale, please re-open if still an issue.

closed 2024-06-04 by h2zero