h2zero/NimBLE-Arduino #489: BLE-HID remote shutter button

doudar, 2022-12-18:
I'm trying to get a remote camera shutter button working with the ESP32. I found some existing code (that looks correct) which finds the remote camera shutter button, connects to it and subscribes to notifications but it doesn't actually receive any notifications. 

The code is here:
https://github.com/esp32beans/BLE_HID_Client/blob/main/examples/BLE_HID_Client/BLE_HID_Client.ino

I slightly modified it so that it would also connect to an "unknown" service with characteristics that advertise notifications in addition to the HID service for more debugging. 

Here's a debug log:
https://pastebin.com/fyc2N7Gj

doudar, 2022-12-18:
Possibly a duplicate of #340 ?

h2zero, 2022-12-18:
Quite likely this is due to multiple characteristics of the same UUID. `getCharacteristic` needs to be extended with an index value to support the cases of multiple same uuids.

doudar, 2023-02-12:
I've narrowed this down to arduino-esp32 version. 1.0.4 works, 2.0.6 doesn't. I'll try to see if something in between works better.

doudar, 2023-02-13:
I had lots of random weirdness going on with this but it seems to be sorted out at least in theory. See #455

closed 2023-02-13 by doudar