h2zero/NimBLE-Arduino #492: NimBLEHIDDevice does not report PNP info and manufacturer string properly

afpineda, 2022-12-30:
**Steps to reproduce**

1. Create a `NimBLEHIDDevice` instance.
2. Call `manufacturer()->setValue()`.
3. Call `pnp()`
4. Start services

Example code:
```c
        hid = new NimBLEHIDDevice(pServer);
        if (!hid) {
            log_e("Unable to create HID device");
            abort();
        }
        hid->manufacturer()->setValue(deviceManufacturer);
        hid->pnp(BLE_VENDOR_SOURCE, BLE_VENDOR_ID, BLE_PRODUCT_ID, PRODUCT_REVISION);
        hid->hidInfo(0x00, 0x01);
        NimBLESecurity *pSecurity = new NimBLESecurity();
        pSecurity->setAuthenticationMode(ESP_LE_AUTH_BOND);
        hid->reportMap((uint8_t *)hid_descriptor, sizeof(hid_descriptor));
        hid->setBatteryLevel(UNKNOWN_BATTERY_LEVEL);
        
        inputGamepad = hid->inputReport(RID_INPUT_GAMEPAD);
        configReport = hid->featureReport(RID_FEATURE_CONFIG);
        capabilitiesReport = hid->featureReport(RID_FEATURE_CAPABILITIES);
        if (!inputGamepad || !configReport || !capabilitiesReport) {
            log_e("Unable to create HID report characteristics");
            abort();
        }
        configReport->setCallbacks(&configFRCallbacks);
        capabilitiesReport->setCallbacks(&capabilitiesFRCallbacks);

        // Start services
        hid->startServices();
```
`BLE_VENDOR_ID` and the other constants are non-zero.

**Expected behavior**

At the hosting PC, the PNP device enumeration should show the given VID and PID, along with the manufacturer string.

**Actual behavior**

The hosting PC show VID=0, PID=0 and an empty manufacturer string.

[image]
[image]

h2zero, 2022-12-31:
I'm unsure what is going on here, I'll have a look into it.

afpineda, 2023-03-25:
Update on this issue:
I found some bugs at `NimBLEHIDDevice.cpp`. Working on a pull request. Most relevant patch is:

```c++
void NimBLEHIDDevice::pnp(uint8_t sig, uint16_t vid, uint16_t pid, uint16_t version)
{
	//uint8_t pnp[] = {sig, (uint8_t)(vid >> 8), (uint8_t)vid, (uint8_t)(pid >> 8), (uint8_t)pid, (uint8_t)(version >> 8), (uint8_t)version};
	uint8_t pnp[] = {
		sig,
		((uint8_t *)&vid)[0],
		((uint8_t *)&vid)[1],
		((uint8_t *)&pid)[0],
		((uint8_t *)&pid)[1],
		((uint8_t *)&version)[0], 
		((uint8_t *)&version)[1]
		};
	m_pnpCharacteristic->setValue(pnp, sizeof(pnp));
}
```

afpineda, 2023-03-25:
Manufacturer string seems to be properly set, from my test with *nRF Connect* on Android. 

I found this post that suggests a bug in the Windows API when retrieving the manufacturer string (which seems to extend to model number and revision strings):
[https://learn.microsoft.com/en-us/answers/questions/401236/hidd-getproductstring-with-ble-hid-device](https://learn.microsoft.com/en-us/answers/questions/401236/hidd-getproductstring-with-ble-hid-device)

h2zero, 2024-06-04:
This should now be resolved.

closed 2024-06-04 by h2zero