h2zero/NimBLE-Arduino #495: ESP32-S3 Stuck on pSvc->getCharacteristic(NimBLEUUID(""));

ILFforever, 2023-01-13:
I'm using an ttgo tdisplay S-3 (https://github.com/Xinyuan-LilyGO/T-Display-S3) powered by an ESP32-S3 chip, trying to connect to a uECG device using the provided code here. (https://github.com/ultimaterobotics/orbitrack_esp32). The code runs and connects fine when ran on an ESP32-Wroom but on the S-3 it freezes on the line  `pChr = pSvc->getCharacteristic(NimBLEUUID("FC7A850D-C1A5-F61F-0DA7-9995621FBD00"));` and returns null; Is this an issue on the ESP32-S3 chip? Furthermore the RSSI returned also is very low at -100

h2zero, 2023-01-20:
At -100 RSSI I would suspect a connection issue. Can you move closer to the device and try?

CurryEx, 2023-01-30:
I had the same problem. If the signal of the BLE device is not stable, for example, blocked by a closing door. Meanwhile, the program already connected to the BLE device and ready to get characteristic, it will likely to be frozen (not everytime) and can not run anymore until trigger the wdt watchdog I set.

closed 2024-06-04 by h2zero