h2zero/NimBLE-Arduino #497: Cannot establish secure connection on ESP32

ohdarling, 2023-01-22:
Hi, I'm using NimBLE to connect ESP32 to a BLE server, and it needs secure connection, but when I perform a secureConnection() on client, it is failed with reason code 13.

I'm testing with the example https://github.com/h2zero/NimBLE-Arduino/blob/release/1.4/examples/NimBLE_Secure_Client/NimBLE_Secure_Client.ino, and just change UUID to expected UUID.

See logs below:

```D NimBLEClient: >> connect(xxxx)
D NimBLEClient: Got Client event BLE_GAP_EVENT_CONNECT 
I NimBLEClient: Connected event
D NimBLEClient: Got Client event BLE_GAP_EVENT_MTU
I NimBLEClient: mtu update event; conn_handle=0 mtu=255
I NimBLEClient: Connection established
D NimBLEClient: >> deleteServices
D NimBLEClient: << deleteServices
D NimBLEClientCallbacks: onConnect: default
D NimBLEClient: << connect()
D NimBLEClient: >> secureConnection()
D NimBLEClient: Got Client event BLE_GAP_EVENT_ENC_CHANGE
E NimBLEClient: secureConnection: failed rc=13
D NimBLEClient: >> getService: uuid: 8000dd00-dd00-ffff-ffff-ffffffffffff
D NimBLEClient: >> retrieveServices
D NimBLEClient: Service Discovered >> status: 7 handle: -1
E NimBLEClient: serviceDiscoveredCB() rc=7 No open connection with the specified handle.
D NimBLEClient: << Service Discovered
E NimBLEClient: Could not retrieve services
D NimBLEClient: Got Client event BLE_GAP_EVENT_DISCONNECT
D NimBLEClient: << getService: not found
I NimBLEClient: disconnect; reason=531, Remote User Terminated Connection
D NimBLEClient: >> deleteServices
D NimBLEClient: << deleteServices
D
 NimBLEClientCallbacks: onDisconnect: default
```

When I use the ESP32 BLE stack, I should use setAuth method of Characteristic with param ESP_GATT_AUTH_REQ_NO_MITM, is there any equivalent code to do that in NimBLE?

Thanks.

**Platform Packages version:**

PLATFORM: Espressif 32 (6.0.0) > ESP32-WROOM-32E (16MB)
HARDWARE: ESP32 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20006.221224 (2.0.6) 
 - tool-esptoolpy @ 1.40400.0 (4.4.0) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5

ohdarling, 2023-01-22:
Also I saw the comment in the header of the example, and performed the erase_flash on ESP32.

h2zero, 2023-02-04:
The error is because the connection times out, which causes the secure connection call to fail and all the subsequent calls will obviously fail also. From the log `rc=13` is `BLE_HS_ETIMEOUT` Operation timed out.

ohdarling, 2023-02-05:
Yes, I think the timeout is caused by the BLE server does not confirmed the connection. When I use internal BLE library,  after setting ESP_GATT_AUTH_REQ_NO_MITM to characteristic, the BLE serve prompts user to confirm the connection, as in the code below:

```
pLocReadChar->setAuth(ESP_GATT_AUTH_REQ_NO_MITM);
pLocReadChar->readValue();
```

After readValue, the BLE server displays a confirmation dialog.

I also tried setSecurityIOCap of NimBLE, but it didn't work:

```
    BLEDevice::setSecurityIOCap(BLE_HS_IO_DISPLAY_YESNO);
    BLEDevice::setSecurityAuth(true, true, true);
```

Is there any equivalent method to do ESP_GATT_AUTH_REQ_NO_MITM in NimBLE?

Thanks.

h2zero, 2023-02-20:
> Is there any equivalent method to do ESP_GATT_AUTH_REQ_NO_MITM in NimBLE?

This should do the same thing: `NimBLEDevice::setSecurityAuth(true, false, true);`

ohdarling, 2023-04-13:
> @ohdarling how did you manage to get these NimBle logs ?

Add build flags in platformio.ini:

```
build_flags =
    -DCONFIG_NIMBLE_CPP_LOG_LEVEL=5
    -DCONFIG_BT_NIMBLE_LOG_LEVEL=0
```

h2zero, 2024-06-04:
closing as stale.

closed 2024-06-04 by h2zero